fossasia/pslab-app #30: Home Fragment UI

akarshan96, 2017-04-06:
[image]

This is a proposed design for Home Fragment UI. The following things will be implemented
- Pinch Zoom
- Improvement in the quality of the image of the PIN Diagram and the description of the PINS.

asitava1998, 2017-04-26:
Add support for a retry/refresh button in case of the app fails to recognize the device or the device is connected after opening the app when device not found is displayed.

viveksb007, 2017-04-26:
@asitava1998 Please open a new issue for this. And tell us if you would like to work on it.

closed 2017-07-18 by viveksb007