fossasia/pslab-app #31: Autogenerate app in apk branch

mariobehling, 2017-04-06:
Please create an apk branch and autogenerate the app here on each build. Please ensure the previous apks are not saved in the history to ensure the repository is not getting super big.

Follow the way it is done for open event android and ask in the open event android channel for help, if needed.

mariobehling, 2017-04-08:
Issue not solved yet.

viveksb007, 2017-04-08:
@mariobehling could you add Authentication token as Environment Variable "GITHUB_API_KEY".
There is authentication problem as can be seen from travis log
[image]

We need Authentication token as GITHUB_API_KEY as can be seen here 
[image]

This should solve the problem.

mariobehling, 2017-04-08:
[image]

Actually you also need a key for the deployment of the branch. 

Please set this up and share the keys with me in private mail.

Cheers.

viveksb007, 2017-04-09:
@mariobehling could you add fossasia github token to travis build settings, my keys won't work as i don't have push access to fossasia's github repo and also change username from wavicles to fossasia in https://github.com/fossasia/pslab-android/blob/master/upload-apk.sh#L13.

Phimpme Android also has fossasia's token in their script as can be seen here.
https://github.com/fossasia/phimpme-android/blob/development/upload-apk.sh#L13

Thanks

viveksb007, 2017-04-10:
@mariobehling  I think we can close this issue as apk in generated in apk branch on any commit to master branch.

PR #34 (2017-04-08): Autogenerate apk in apk branch
Fixes #31 should work
and updated readme.md showing setup to use PSLab with Android App
closed 2017-04-10 by mariobehling