fossasia/pslab-app #32: Outline development directions and goals

mariobehling, 2017-04-07:
A discussion about the Android app has been taking place in the pslab issue here: https://github.com/fossasia/pslab/issues/35

First python code has been ported to Android. Please outline the direction and goals of the development that has been started here.

viveksb007, 2017-04-07:
Python Communication library has been ported ( ~60% ). Testing of that ported code needs to be done.
Once successful in communication with PSLab device we can map the functionality with current UI and discuss and develop UI for more functionality.  

There is the main Navigation Drawer : 

[image]

Here **Home Fragment** tells user about status of device connected and details about device like version number, product ID, etc. And there would be a PIN Diagram along with PIN description of PSLab device.
### Home Fragment
[image]

In **Applications**, current UI proposed is this https://github.com/fossasia/pslab-android/issues/30 .

we need UI to expose the PSLab functionality mentioned in image below : 

[image]

Saved Experiment and Design Experiment would be worked on once Applications part is functional.

Other fields in NavDrawer are self explanatory.
 
@jithinbp @wavicles @akarshan96 Any addition or changes are welcome.

akarshan96, 2017-04-07:
The following approach (as mentioned by @viveksb007)  is I think to be followed. In a nutshell, development will take place the following way:
- Porting Python Communication ported to Android.
- Testing and Debugging of the ported code.
- Developing UI for each application as mentioned by @viveksb007 and linking UI components with methods of ScienceLab.java in order to expose PSLab functionalities (Like now, we will open an issue proposing an UI design.)
- Testing and Debugging each PSLab Application we create.
- Integrating it with the main code.

mariobehling, 2017-04-07:
Please state which prerequisites are necessary in order for this Android app to function (and add this to the Readme.md). There were two approaches outlined in https://github.com/fossasia/pslab-python/issues/35#issuecomment-286819978:

> A) port the python communication library as well as apps to Java . This will enable running PSLab on OTG enabled android phones (most modern phones have OTG support) The device needs to be physically connected to the phone via a cable.

> B) utilise the provision in the hardware to add an ESP8266 WiFi module for communication . The current footprint is for an ESP12E module.The ESP module can run a lightweight web server

I did not see a final decision being taken. The outcome of the discussion should be clearly stated and added to all relevant Readme.md files in the different repositories. If the decision is to go with an ESP/server approach, please add this info and close the issue in the other repository.

closed 2017-08-13 by viveksb007