fossasia/pslab-app #33: Oscilloscope UI

akarshan96, 2017-04-07:
This is the first step towards the development Oscilloscope's UI.

[image]

[image]
The above image showcase the landscape mode of the Oscilloscope Activity. 
Gestures will also be integrated, like
-pinch zoom for changing the timebase.
-double tap on the graph to freeze.
-swipe down to save.

Please give suggestions on:
-layout and what needs to be improved.
-real-time graph api to be used ( I have used GraphView http://www.android-graphview.org). We can also use MPAndroidChart https://github.com/PhilJay/MPAndroidChart).
-I have researched on 3D graphs libraries but they are paid.
-any other thing I missed.

akarshan96, 2017-04-09:
[image]

This is the screenshot of Oscilloscope Activity running on a 7inch tablet.

asitava1998, 2017-04-26:
I think we can add the various types of trigger modes found in the oscilloscopes used in Labs. They are as follows:-
A. Auto -  This trigger mode allows the oscilloscope to acquire a waveform even when it does not detect a trigger condition. If no trigger condition occurs while the oscilloscope waits for a specific period (as determined by the time-base setting), it will force itself to trigger.
B. Normal -  The Normal mode allows the oscilloscope to acquire a waveform only when it is triggered. If no trigger occurs, the oscilloscope will not acquire a new waveform, and the previous waveform, if any, will remain on the display.
C. Single -  The Single mode allows the oscilloscope to acquire one waveform each time you press the RUN button, and the trigger condition is detected.
D. Scan – The Scan mode continuously sweeps waveform from left to right.
I can implement these in the oscilloscope UI and these would help to make it more elaborate.

Another feature that I would like to suggest is automatic scaling of the graph. Suppose the graph has axes set between 0-100 units and then keeps receiving data between 0-50 units, we can automatically scale up the graph for better view instead of the user zooming into it.

akarshan96, 2017-05-07:
@asitava1998 and I come up with an idea to have a bar at the bottom of the screen which shows the status of the device and other information like sampling rates etc.

images indicate that the PSLab is connected or disconnected
[image]
[image]

PR #149 (2017-06-13): Created Oscilloscope UI
fixes #33 
Sorry for the long PR, it might be difficult to review. From next time I will break the issue into sub-issues.
video link - https://goo.gl/photos/mdbNoaGZHk8MBoreA
Screenshots
[image]

**Channel Parameters**

[image]

**Timebase and Tigger**

[image]

**Data Analysis**

[image]

**XY Plot**
closed 2017-06-13 by viveksb007