fossasia/pslab-app #41: Control Activity's UI

akarshan96, 2017-04-11:
The following are the screenshots of proposed design of Control Activity.

[image]

[image]

The following has been implemented using 2 ListViews.
Please give suggestions and also review Oscilloscope's UI #33 so that I can make amendments while the python code is been ported.

jithinbp, 2017-04-11:
This looks neat. :)
A few thoughts :
1)  Presently the labels are displaying the value that the user is trying to set via the slider. In practice , it should display the actual value that was set after considering resolution limitations . e.g. setting 3.001 V on PV1 (+/-5V @ 12-bit  => ~2.5mV step size ) might actually set 3.0 or 3.002 , so the correct value may be shown. The set value is returned by the set_pvx functions. This is especially important at the boundaries of the voltage ranges where op-amp offsets can result in differences as high as 30mV. 

2) The voltmeter read function stops any active oscilloscope capture routines because the ADC is re-initialized to read 12-bit values with 16x averaging. Therefore, if the activity tab is provided in parallel, the oscilloscope UI must be prepared to handle a voltage read event.

I propose to merge the Low frequency and High frequency function calls. The comm lib can automatically invoke high_frequency if the frequency measured by low_frequency is greater than around 100KHz

Can you please put up the instructions to build the Apk? Tools, dependencies etc
I've got Android Studio 2.3.1 with JDK 1.8.0_112 . It then proceeded to download gradle 3-3 . Now I'm getting build errors.
Thanks

akarshan96, 2017-04-11:
Thanks, @jithinbp for the review.
- 1 and merging of the low frequency and high frequency into a common library will be implemented as soon as sciencelab.py is ported. 
- I am now working on 2.

wavicles, 2017-04-11:
@akarshan96  Awesome :+1:

viveksb007, 2017-04-13:
@akarshan96 These won't be implemented using ListViews, we generally use ListView when we need a trigger on click of complete row item, like i explained in Application Fragment https://github.com/fossasia/pslab-android/pull/29. In this case we need individual OnClick and OnSelect, so refrain from using ListView ( We can do it inside custom adapter, but it would be quite messy ).

And why are we reading values inside Control Activity?

If I got it right, when we open Control Activity, values of PV1,PCS, etc. will be set to current values of PSLab device, and we can change them( taking resolution in consideration ) and there should be a Save/Apply button at bottom of list to force change the values of PSLab device.

And I think we should move reading Voltage,Capacitance, etc to a new item in List of Application Fragment.  And two Lists on single mobile screen might be a poor  UX. @jithinbp @wavicles what do you think?

jithinbp, 2017-04-13:
@viveksb007 I'm no expert , but I would lean towards housing the widgets in a ScrollView if it saves complications. If combinations of widgets are involved ( slider + label ..) , a linearLayout may be used for each. Maybe we need someone with more experience to weigh in on this, but it doesn't sound like a big deal either ways.  2 consecutive listviews might be an annoyance depending on how they handle scroll events.

Regarding the second query , I feel it's more practical to have all direct-access functions in one tab. Maybe we could rename it to ' Test and Measurement ' . In the desktop app it was called 'controls' owing to space constraints in vertical tabs.
Why I feel they should be together, is because users will often want to read-back the consequences/outcomes of some value they set, and placing them in separate tabs will make it tedious to switch back and forth.
For example, a student can quickly acquire a set of datapoints to verify a DC summing junction circuit by using PV1,PV2, and CH1

akarshan96, 2017-04-13:
@viveksb007 @jithinbp I have added 2 listviews in a single screen to give the user an idea of different PSLab functionalities they can access via the Control Activity. These lists dynamically change according to the screen size. I know using onClick inside Custom Adapters is little tricky, but I implemented it to make UI neat. I can definitely change ListViews to LinearLayouts.

Let me propose a new design keeping in mind above points in discussions. The below is a basic idea of a new design we can implement.

[image]

[image]

This activity uses tabs/buttons to switch between control and read events and  a single screen can be used by each list.

Please tell if you want me to make amendments in the previous approach or work on a design I proposed above or think of another design.

viveksb007, 2017-04-13:
@jithinbp got the usecase.
@akarshan96 Use tabstrip layout with fragments.
Put settings values part in one tab and reading in another. switching between them is simple sliding.

[image]

UseTabStrip with ViewPager and follow material guidelines while making UI.

akarshan96, 2017-04-13:
@jithinbp @wavicles @viveksb007 This is a design of Control Activity using tabs. I have used layouts instead of listviews. 

[image]

[image]

commit 07e35bc2 (2017-06-13): Fixes #41 creates UI for Control (#154)

* Fixes #41 creates UI for Control
closed 2017-06-13 by viveksb007