fossasia/pslab-app #48: Communication error

jithinbp, 2017-04-14:
To be tested with hardware
```
java.lang.RuntimeException: Unable to start activity ComponentInfo{org.fossasia.pslab/org.fossasia.pslab.activity.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.hardware.usb.UsbDeviceConnection.claimInterface(android.hardware.usb.UsbInterface, boolean)' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2423)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483)
	at android.app.ActivityThread.access$900(ActivityThread.java:153)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:148)
	at android.app.ActivityThread.main(ActivityThread.java:5441)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.hardware.usb.UsbDeviceConnection.claimInterface(android.hardware.usb.UsbInterface, boolean)' on a null object reference
	at org.fossasia.pslab.communication.CommunicationHandler.open(CommunicationHandler.java:80)
	at org.fossasia.pslab.communication.ScienceLab.<init>(ScienceLab.java:34)
	at org.fossasia.pslab.activity.MainActivity.onCreate(MainActivity.java:74)
	at android.app.Activity.performCreate(Activity.java:6303)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1108)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2376)
	... 9 more

```

viveksb007, 2017-04-14:
expected as code wasn't tested.
will try to solve communication error as soon as i get the device.
@jithinbp I took reference from your comm. library and https://github.com/mik3y/usb-serial-for-android.

akarshan96, 2017-04-14:
@viveksb007 I think the Androidmanifest.xml should be configured to notify your app when a device is attached.

jithinbp, 2017-04-14:
@akarshan96 yea, I remember using those to automatically launch the App when the device was connected. Another way would be to ask for permission when the app is launched. The following code was used by the MainMenuActivity in the [Android app for ExpEYES](https://github.com/jithinbp/expeyes_android_library)
```
private static final String ACTION_USB_PERMISSION = "com.android.example.USB_PERMISSION";
private PendingIntent mPermissionIntent;
	@Override
	public void onCreate(Bundle savedInstanceState) {
		super.onCreate(savedInstanceState);
		setContentView(R.layout.activity_main_menu);
		
		setTitle("ExpEYES Experiments ( Connect... )");
		
 		mPermissionIntent = PendingIntent.getBroadcast(this, 0, new Intent(ACTION_USB_PERMISSION), 0);
		filter = new IntentFilter(ACTION_USB_PERMISSION);
		registerReceiver(mUsbReceiver, filter);
        
        mUsbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
        mcp2200 = new devhandler(mUsbManager);
        
        if(mcp2200.device_found )mUsbManager.requestPermission(mcp2200.mDevice, mPermissionIntent);
        else Toast.makeText(getBaseContext(),"No device connected. check connections.",Toast.LENGTH_SHORT).show();
        ej=expeyesCommon.getInstance();
        
        
        
	}
```

jithinbp, 2017-04-14:
Look for
```
/*---------------------REQUEST USB PERMISSION WITHIN THE APPLICATION--------------------------*/
```
in (https://github.com/jithinbp/ExpEYES_android/blob/master/src/com/expeyesexperiments/MainMenuActivity.java)

akarshan96, 2017-04-14:
@jithinbp yes, we have to define a onReceive method under BroadcastReceiver. I am working on it. Hope it resolve errors

viveksb007, 2017-04-14:
@akarshan96 this onReceive method is for explicitly asking for permission when there are more than one device connected.
Using an intent filter should solve the issue.
Reference : https://developer.android.com/guide/topics/connectivity/usb/host.html#permission-d

PR #62 (2017-04-21): Communication Error Fixed
Fixes #48 
and updated version string in HomeFragment after communicating with device.

[image]
commit c3f756ec (2017-04-21): Communication Error Fixed (#62)

* Fixes #46 build instructions for standalone apk

Added build instructions for standalone apk and debug over Wi-Fi

* Fixes #48 communication error fixed
closed 2017-04-21 by wavicles