fossasia/pslab-app #51: faulty polynomial implementation in DACChannel.java

viveksb007, 2017-04-16:
In DACChannel.java coefficients for 0 & 1 are put [see here](https://github.com/fossasia/pslab-android/blob/master/app/src/main/java/org/fossasia/pslab/communication/peripherals/DACChannel.java#L29)
And in MCP4728 coefficient for 2 is queried [see here](https://github.com/fossasia/pslab-android/blob/master/app/src/main/java/org/fossasia/pslab/communication/peripherals/MCP4728.java#L63), as its put in a map, it would return null and null in mathematical operation would raise error.

akarshan96, 2017-04-16:
@viveksb007 thanks. I got that, going to resolve it.

keshcode, 2017-04-16:
i would like work on this issue

viveksb007, 2017-04-16:
@keshcode thanks for showing interest in this project.
Feel free to suggest or implement ways in which we can implement polynomial efficiently in java ( numpy.poly1d() in python ).
You can discuss your approach at our gitter channel https://gitter.im/fossasia/pslab
The python communication library is at https://github.com/fossasia/pslab-python

akarshan96, 2017-04-17:
@viveksb007 I have solved the issue but when I am making a pr, the previous commits which are already merged are also being displayed there. I have tried solutions mentioned on the net but still facing this issue. Should I still make a pr ?

viveksb007, 2017-04-17:
Its most probably because wavicles might have rebased the commits.
As the changes are quite less, you can edit that on github only. would save duplicate commit.

akarshan96, 2017-04-17:
@viveksb007 Thanks.

PR #60 (2017-04-19): Updated MCP4728.java, Ported some Sensor files 
- Updated  MCP4728.java to fix #51 
- Ported some Sensor files
-Added TSL2561.java #53 
-Added MLX90614.java #57 
-Added SHT21.java #54 
-Added BMP180.java #56
closed 2017-04-19 by wavicles