fossasia/pslab-app #71: Polynomial implementation approach revisited

akarshan96, 2017-04-27:
Apache Math Common provides a range [Polynomial functions](http://commons.apache.org/proper/commons-math/javadocs/api-3.6.1/index.html) that can be used to implement polynomials efficiently.

viveksb007, 2017-04-28:
Great we should switch to Apache Common Math. Just take care of one thing about evaluation in numpy.poly1D  polynomial is solved as 
`poly1d([1, 2, 3])  -> x^2 + 2x + 3`
and in Apache Common Math, its evaluated as
`3X^2+2X+1`
So reverse of coefficients should be given when constructing a polynomial instance in Apache common math.

viveksb007, 2017-05-07:
@akarshan96 when can you incorporate Polynomial implementation from apache common math?
#91 Could only be addressed after underlying polynomial implementation is done.

PR #99 (2017-05-08): Updated files according to new Polynomial appraoch
Modified Polynomial implementation approach. Fixes #71
closed 2017-05-08 by viveksb007