fossasia/pslab-app #72: Handling abrupt disconnection/device not found cases

asitava1998, 2017-04-27:
Addition of a manual Retry/Reconnect button in the Home UI if connection fails even when the device is connected.
While in operation, if the device gets abruptly disconnected, two solutions can be done-
1. Landing back on Home UI and trying to reconnect again.
2. Displaying a toast asking user to manually reconnect again.

Although its a minor issue, I would like to do it to get started and also get a hang of this project.

viveksb007, 2017-04-28:
@asitava1998 Great, start working on it and send a PR.
I was thinking, we should have a reconnect icon on the actionBar when UI is HomeFragment only or refresh by pulling HomeFragment down.

viveksb007, 2017-06-20:
@asitava1998 has there been any improvement for solving this issue?

asitava1998, 2017-06-20:
@viveksb007  I had added a button to refresh the connection in this issue. However, after that changes were made to how the sciencelab class works (provides instances), I still have the code with me, can make a few changes in it and send a PR.
However, swipe to refresh cannot be implemented in the home fragment as it is limited to ListView and RecyclerView and the home fragment does not use these views.

viveksb007, 2017-06-22:
@asitava1998 saw your scrum, I already worked on it yesterday. And after testing would send a PR today.

viveksb007, 2017-07-18:
Fixed in #217

closed 2017-07-18 by viveksb007