fossasia/pslab-app #91: modify polynomial processing functions in AnalogChannel

viveksb007, 2017-05-06:
modify polynomial processing functions in AnalogInputSource to handle arguments as single value and array both.
Source File [link](https://github.com/fossasia/pslab-android/blob/master/app/src/main/java/org/fossasia/pslab/communication/analogChannel/AnalogInputSource.java)

New Polynomial Implementation should be taken into consideration i.e Polynomial object provided by apache common math library and necessary changes should be made.
About Apache Common Math Polynomial https://github.com/fossasia/pslab-android/issues/71

akarshan96, 2017-05-07:
@viveksb007 I have modified entire repo implementing with the new Polynomial approach using Apache Maths Commons. That includes AnalogInputSource.java. Should I make a pull request or drop the commits as I see you already working on it?

viveksb007, 2017-05-07:
Make a PR. 
And have you modified functions accordingly to find value of polynomial at different data points?

akarshan96, 2017-05-07:
@viveksb007 I have made the changes in the functions too, let me make a PR and you can review and point if something I missed.

PR #87 (2017-05-08): some capture routines added in sciencelab
Fixes #91 
Partly fixes #86 
Rest of capture routines would be added as soon as some implementation details is clear in python-communication library.
closed 2017-05-08 by wavicles