fossasia/pslab-app #92: Changing "Applications" list with cardviews

CloudyPadmal, 2017-05-06:
The current _Applications_ view consists of a list view with basic list items. It doesn't have much of a description on what each item would do.

| Current View | 
| ------------- | 
| [image] |

This view can be enhanced with the help of a list view with card items or a grid view as follows!

| Card Views | Grid Views |
| ---------------| -------------- |
| [image] | [image] |

If this feature is okay to implement, I can get to work on it! :)

viveksb007, 2017-05-06:
Could you tell a bit about what description you want to show?
I think the name are self explanatory.
Could you look into when the application fragment is rendered for first time, the transition isn't so smooth?
If you are facing the same problem.

CloudyPadmal, 2017-05-06:
Yes @viveksb007 it lags a bit! I can work on it!! :+1:

asitava1998, 2017-05-06:
I think you should implement the GridViewLayout. It would be more compact as scrolling through the layout is less and there is no need of descriptions as Vivek mentioned.

CloudyPadmal, 2017-05-06:
@asitava1998 I'll implement a GridView and send a PR! :) Currently I'm trying to fix the lag happening when _Applications_ view is opened for the first time!

akarshan96, 2017-05-06:
What I am thinking is to have both listview and gridview. Provide a user button that would convert the list to grid and vice versa. The button will make one view visible while another invisible. What say ?

CloudyPadmal, 2017-05-06:
That would be cool right? :D May be in settings we can provide an option to select which view user needs!

CloudyPadmal, 2017-05-07:
How's this UI? :) 

[image]

asitava1998, 2017-05-07:
Looks neat to me.

CloudyPadmal, 2017-05-07:
Thank you @asitava1998 !! Made the titles Bold!

[image]

akarshan96, 2017-05-07:
@CloudyPadmal Looks great. One suggestion, we can add round corners to each tile. What say?

jithinbp, 2017-05-07:
I think we need a thorough discussion on the UI . Can we all spend some
time on some online collaborative UI editor?

I feel the controls/advanced controls section should be accessible in the
form of a slide-out drawer regardless of what experiment is running. That
way it's more like having separate instruments at hand .

Of course, one may have control-widgets pertinent to each experiment in its
own layout also, but the rest should also stay within reach

asitava1998, 2017-05-07:
@jithin That would be better as so far not all of us have been involved in giving suggestions about the UI. Mario suggested the use of [Invision](https://www.invisionapp.com), we can try this one as it supports collaboration or any other that others might suggest.

CloudyPadmal, 2017-05-08:
@akarshan96 that would be nice right? I'll add!! @jithinbp that's a good idea! I agree! :+1: Let's try Invision !!

commit 88d53064 (2017-05-09): Fixes #92 : Applications view UI redesign! (#106)
closed 2017-05-09 by viveksb007