fossasia/pslab-app #93: Opening "Applications" window for the first time lags!

CloudyPadmal, 2017-05-06:
As @viveksb007 pointed out, the application lags a little bit when we're opening the _Applications_ tab for the first time.
As observed from the logs, 
> 05-06 22:53:11.275 8299-8299/org.fossasia.pslab D/dalvikvm: GC_FOR_ALLOC freed 2460K, 14% free 22474K/25992K, paused 35ms, total 39ms
05-06 22:53:11.275 8299-8299/org.fossasia.pslab I/dalvikvm-heap: Grow heap (frag case) to 24.156MB for 1440016-byte allocation
05-06 22:53:11.305 8299-8308/org.fossasia.pslab D/dalvikvm: GC_FOR_ALLOC freed 6690K, 38% free 17190K/27400K, paused 22ms, total 22ms
05-06 22:53:11.325 8299-8299/org.fossasia.pslab D/dalvikvm: GC_FOR_ALLOC freed 62K, 38% free 17127K/27400K, paused 16ms, total 16ms
05-06 22:53:11.325 8299-8299/org.fossasia.pslab I/dalvikvm-heap: Grow heap (frag case) to 20.651MB for 3240016-byte allocation
05-06 22:53:11.356 8299-8308/org.fossasia.pslab E/dalvikvm: adjustAdaptiveCoef max=4194304, min=1048576, ut=568
05-06 22:53:11.356 8299-8308/org.fossasia.pslab D/dalvikvm: GC_FOR_ALLOC freed 0K, 26% free 20291K/27400K, paused 25ms, total 25ms
05-06 22:53:11.386 8299-8303/org.fossasia.pslab E/dalvikvm: adjustAdaptiveCoef max=6291456, min=1572864, ut=368
05-06 22:53:11.386 8299-8303/org.fossasia.pslab D/dalvikvm: GC_CONCURRENT freed 1K, 14% free 23625K/27400K, paused 6ms+4ms, total 28ms

there is a possible memory intensive task is happening. Currently I am investigating on this issue!

akarshan96, 2017-05-06:
I think this lag is due to the listview is loading a lot of images. I think reducing sizes of the images will do the trick.

CloudyPadmal, 2017-05-06:
It seems so @akarshan96 !!

akarshan96, 2017-05-06:
@CloudyPadmal does it still lags?

CloudyPadmal, 2017-05-06:
Now it's fine after image size reduction! :) :+1:

commit ea32b2b6 (2017-05-07): Fixes #93 : Image sizes reduced! (#94)
closed 2017-05-07 by viveksb007