fossasia/pslab-app #96: Create a list of features of each screen

mariobehling, 2017-05-07:
Please create a list of features in the Readme.md and add the status "implemented", "under development" etc.

asitava1998, 2017-05-07:
## Oscilloscope UI
**Features to be included:**
* Channel I/O plots
* X-Y plots
* Trigger modes (Auto, Normal, Single, Scan)
* Setting Scale
* Freeze & save graph, pinch zoom
* Find duty cycle, maxima & minima, offset calculation
* TimeBase
* Sine fit, Fourier Transform
* Overlay
* Display sampling rate, vertical sensitivity, separate display of DC & AC parts ( check feasibility of implementation )

**APIs to be used:**
* Graphs: GraphView and PhilJay
* Sin fit, Fourier Transform
* Apache Maths Commons

**Notes:**
* ScrollView will be implemented in Oscilloscope UI. The ScrollView will be implemented below the   Graph so that user can access various functionalities and control of Oscilloscope without the Graph  being dissapear.
* Step size of timebase and trigger should be taken in consideration.

**Color Schemes:**
* Red, Grey, White (As they match the Organization’s theme)

asitava1998, 2017-05-07:
##Control UI
* Features to be include:
* Voltmeter (measure voltage)
* Measure Resistance
* Measure Capacitance
* PCS, PV1, PV2, PV3, W1 (get their readings)
* Measure Frequency
* Measure/Count Pulses

CloudyPadmal, 2018-06-21:
Features added in `README.md` and updated their status on #1105

closed 2018-06-21 by CloudyPadmal