fossasia/pslab-app #107: Incomplete code at DigitalChannel.java

CloudyPadmal, 2017-05-09:
The method `loadData` in the `DigitalChannel.java` file contains an incomplete if else statement at lines 54-60
```Java
void loadData(int initialState, double[] timestamps) {
    if (initialStateOverride) {
        // Code not clear
        initialStateOverride = false;
    } else {

    }
```

viveksb007, 2017-05-09:
Yeah it was not clear at time. Well its still a little ambiguous.
https://github.com/fossasia/pslab-python/blob/development/PSL/digital_channel.py#L41
Have a look at this, initial_state_override was declared a boolean. So whats happening in this line?
If you could decode it. Complete the method a send a PR please.

CloudyPadmal, 2017-05-09:
Yes @viveksb007 ! It's a little confusing! In python, anything other than 0 is evaluated true. Not sure what is evaluated in this line of code. I'll give it a try and see 😊

commit 6af91b70 (2017-05-11): Fixes #107 : DigitalChannel loadData method completed! (#113)
closed 2017-05-11 by viveksb007