fossasia/pslab-app #108: Define detailed project goals and develop user stories - What can a user do with this app?

mariobehling, 2017-05-09:
There has been a good start developing the app and developers have started thinking about screens. However, it is unclear why certain buttons, screens and features are added. The reason why the app has certain features seems unclear and it has not been formulated what this app is actually trying to achieve in detail.

We need a reference document to understand what this app is good for specifically apart from the general idea to help us to do experiments.

Let's proceed to a more structured approach therefore. It is your job to think this through in a logical way, what is the goal of the app and every single feature in the app. Developers are the first users and the first testers of their own ideas!

We need a list of features depending on what the user can do (and wants to do). So my question is without thinking about the interface, please tell us: 
* What do you want to do with this app? (I am not looking for an answer "I want to do experiments.". I would like to know what do you want to do exactly, e.g. connect devices (which ones and how, what information do you need etc.).
* What do you want to find out using this app, what do you want to measure, why and how do you want to do it with this app and device etc.
* Why do you want to install this app and buy PSLab potentially? How does it help you? (Again not looking for general answers like "It can help me to understand Physics.") 
* Please tell us specifically what this app and device should do for you, that you cannot do without it.

Based on your answer, we can define a list of features and priorities.

Please create a Google document (each developer) and answer these questions on at least one A4 page and use the concept of user stories.

@viveksb007 @jithinbp @CloudyPadmal @akarshan96 @asitava1998

asitava1998, 2017-05-09:
I have listed my answers/views [here](https://docs.google.com/document/d/1C2UcyVSkvECzq6SGKLeqpUqSgrO-8YXTTac_P4sGmeQ/edit)

CloudyPadmal, 2017-05-09:
User stories generated by me! [Why PSLab??](https://docs.google.com/document/d/1OxjYCXg3x95nQL7saIOngjfLlx6B81Dz2k8B8CyeW-Q/edit?usp=sharing)

jithinbp, 2017-05-09:
Common user stories and expectations from the app : My take on  Why PSLab?
<https://docs.google.com/document/d/1VI7shZpG8zAIcWHRt1db0sb7JEf1XuN0vSDCWxuEtrU/edit?usp=sharing>

akarshan96, 2017-05-10:
I have written my answers [here](https://docs.google.com/document/d/1qO8fgeuOxFuRmng5-2i6u_rstQ5lx7-PcorSXWBIj_o/edit?usp=sharing).

viveksb007, 2017-05-10:
My Views on usability of Android App for PSLab device [gdocs](https://docs.google.com/document/d/1Jk2fkd8vUyi1m7RGfuhzg1r0xnHEOsqh-RkIG0qWi5U/edit?usp=sharing)

CloudyPadmal, 2018-07-16:
Related to GSoC 2017

closed 2018-07-16 by CloudyPadmal