fossasia/pslab-app #120: Reduction of main application icon sizes in landscape mode

jithinbp, 2017-05-17:
I tried out the app on a Xiaomi Redmi 3S, and the auto-detection is working great.
I would like to suggest a slight tweak regarding the grid icon sizes in landscape mode. They look needlessly massive on the 5" 720p display. A three column format might feel better than the two column one currently used
[image]
[image]

viveksb007, 2017-05-20:
@CloudyPadmal could you have a look at this too?

CloudyPadmal, 2017-05-20:
Oh this looks terrible! Sure @viveksb007 ! I'll get into this!! 👍

commit 63de9a1e (2017-05-24): Fix #120 : Applications grid colomns change in landscape orientation (#127)
closed 2017-05-24 by viveksb007