fossasia/pslab-app #121: Error in reading complete data from PSLab device

viveksb007, 2017-05-18:
I tried reading 2048 bytes from Pslab device and was only able to read 1889 bytes. 
I tried many times but every time I was reading 1889 bytes after that error occurred.

[image]

viveksb007, 2017-05-18:
@akarshan96 could you confirm this with the device that you have?
Just take a scienceLab instance from sciencelab common and read 2048 bytes Flash Data at page 3.

jithinbp, 2017-05-18:
This can be due to small buffer sizes/too many concurrent processes. I have observed this on a Raspberry Pi. The solution there was to lower the BUAD rate, but that's a hack.
Instead of read_bulk_flash,  you can also try read_flash
read_flash reads in chunks of 16 bytes, where each page is segregated into 2048/16 locations .
So,
This - `read_bulk_flash(page1,2048)`
becomes
```
a=[]
for b in range(2048/16):
    a+=read_flash(page1,b*16) # read 0-15,16-31... and add them all
```

viveksb007, 2017-05-18:
Okay I will try by reading small chunks and adding them after.
Is there any way to increase fifo buffer size? And what is the current buffer size?

jithinbp, 2017-05-18:
It's not a buffer problem of the device.
It's a buffer issue with the Host. RPi had only 64 bytes of RX buffer, so
it tended to overflow if the software didn't empty it out in time.
Android devices might also have the same issue, but I'm sure there's a
software based fix.
for now I think the read_flash routine should work just as well. After all,
It's only required to be called during initialization. Even if I split the
read_bulk_flash into segmented reads, you'll still be doing multiple
transactions.

On a related note : Once we move to using only polynomial based fitting,
the net calibration chunk will be less than a hundred bytes. the current
routine where entire DAC tables and an INL table of the ADC is stored in
massive arrays is overkill for this hardware.

viveksb007, 2017-05-19:
@jithinbp doing read_flash in a loop is taking a lot of time, to be precise approx 64sec. Did it take this much time on RasPi too?

jithinbp, 2017-05-19:
Shouldn't. Maybe you're encountering a timeout somewhere. How much time
does it take for a single read_flash?

On raspberry pi, I used the set_baud function to lower the baud rate on the
hardware, and then reopened the serial port with the new baud. Was
cumbersome, so I eventually switched to Orange pi pc, and stuck with it.

viveksb007, 2017-05-19:
single read_flash takes 500ms.
When can we move to polynomial based fitting and why can't we add calibrated data in some configuration file in app itself instead of reading it from device?

jithinbp, 2017-05-19:
500mS for 16+1bytes? Seems something isn't right with the function call.
Using the Python library, reading 100 locations(1600 bytes) from page 3
took 130mS

Each device carries its own unique calibration data. What one can do is
read the calibration data and save it to a local file along with the device
ID so that it doesn't have to be read on each launch of the app.

viveksb007, 2017-05-20:
[image]

successfully read, bug was error in reading acknowledgement, I wasn't able to read 1 byte from endpoint. So it must be going to some interrupt sub-routine to handle error and that took time.
So read 17 together. Now it took 854ms to read 2176 bytes. Not an elegant way but works.
Related to why #79 wasn't solved.
Would dig more to find why this is happening.

PR #131 (2017-05-29): Updated calibration method 
Fixes #121 Successfully read complete calibration data in chunks.
Fixes #129 Unpacked Correctly
Partly Fixes #104 ( creating a Polynomial Function from read values is remaining )
All the above issues were from same method, so updated them in single PR.
commit 6b54700b (2017-05-29): Updated calibration method  (#131)

* fix #121 successfully read complete calibration data

* fixes #129 unpacked float correctly

* Fixes #104 partly fixes calibration method

* fixed build error

* fixed error in unpacking float
closed 2017-05-29 by mariobehling