fossasia/pslab-app #139: Add template for Issue and PR

viveksb007, 2017-06-01:
Add template for issue and PR as other projects like [susi_android](https://github.com/fossasia/susi_android/tree/development/.github),etc this gives a format to raise issues and submit a PR.

CloudyPadmal, 2017-06-04:
Shall we create a format like the one in Susi? They all share the same format it seems! What do you think?

viveksb007, 2017-06-04:
Our project isn't in that stage where we can use the same format. We can delay this for now.

viveksb007, 2017-06-13:
@CloudyPadmal I think we should add this template as we have started working with UI and screenshots for describing the Issue and screenshot for showing the changes in PR would be great.
I am adding susi_android template. Is that okay with you?

CloudyPadmal, 2017-06-13:
@viveksb007 yes that would be great! I checked their template and it will suit us well! :+1: :100:

PR #163 (2017-06-13): added template for ISSUE and PR
Fixes #139 
Added template for Issue and PR and now we have started working with UI. So a proper structure to report issue and submit PR would be great.
Copied susi_android template for now.
closed 2017-06-13 by viveksb007