fossasia/pslab-app #142: Logic Analyzer UI

viveksb007, 2017-06-04:
Coding UI for Logic Analyzer by taking reference from [LAmockup](https://app.moqups.com/viveksbhadauria007@gmail.com/Xz2hJPtEc5/view) here.

viveksb007, 2017-06-11:
[image]

Used different charts for different logic lines.
Here is a link for video that I made by changing parameters. Would add a slider panel to change input channel and edge select on Logic Lines Screen. Would look into plotting logic line as ideal logic line i.e both high and low at same time during transition. Its default plotting of MPAndroidChart.

https://goo.gl/photos/axoymdEgzHQBnyFK7

@jithinbp @akarshan96 @asitava1998 @CloudyPadmal have a look and provide review comments. Thanks

jithinbp, 2017-06-11:
@viveksb007 thank you. My thoughts : 
-  It's a lot easier to view multi-channel logic signals if the X-Axes are linked together. Scaling the time axis of one plot, while the others stay the same is misleading.
-  I see you've used a sample dataset that changes the level periodically. Issues that should be noted:
    - The Logic Analyzer functions only tell you when the logic level changed. There are no datapoints if the level did not change. 
    - For each level change, you need to generate two datapoints to draw a vertical line:[ (timestamp, Y_OFFSET), (timestamp,Y_OFFSET+height_of_channel ) ] or vice versa .   Y_OFFSET depends on the channel number  .   If you ported digital_channel.py completely , this has already been done.

Here's a screenshot of the implementation in pslab-desktop-apps. It's an early revision, so ignore the axis labels and stuff. But otherwise it conveys a clear picture of the logic states, and zooming along the X direction will easily reveal time-critical details.
[image]

And here's a screenshot from a commercial app. Looks like Saleae's app, but I'm not sure.

[image]

viveksb007, 2017-06-11:
@jithinbp Thanks for the review. 
So to solve this I am thinking of using a single chart and clamping signals from different channels at different y-offsets. Need to take care of labels on Y-axis. Would that be okay?
And when logic analyzer would return a list of timestamps and initial state, I would generate data points accordingly. This data-point sample was generate just to show plotting of logic lines.
Interchanging of stacking of charts might be difficult, but would look into it after basic functionality is done.
I will put updated screenshots as soon as its done.

viveksb007, 2017-06-11:
I have made the following changes:

- Plotted Ideal Logic Lines
- All plots are on same chart so, time axis change is same in all channels simultaneously
- Generated random timestamps for plotting data.

ToDo 

- Need to take care of Labels on Y-Axis.

Link for new ScreenCast : https://goo.gl/photos/j31ZZvy9AN3LQzt47
@jithinbp could you have another look please?
Thanks

PR #155 (2017-06-13): Created Logic Analyzer UI
Fixes #142 
All logic lines have been created on a single chart.
- Plotted Ideal Logic Lines
- All plots are on same chart so, time axis change is same in all channels simultaneously
- Generated random timestamps for plotting data ( As a test data ).

Link for screencast : https://goo.gl/photos/j31ZZvy9AN3LQzt47

[image]
closed 2017-06-13 by viveksb007