fossasia/pslab-app #144: Naming convention for Views

asitava1998, 2017-06-09:
Since, each developer is working on different section of PSLab. It is important that a naming convention is followed to maintain uniformity in the code. For example one possible convention can be                                    
(viewtype)_(activity/fragment name)_(no. in the file).xml   like button_activity_oscilloscope.xml.

viveksb007, 2017-06-09:
Yeah that okay, I have been using like this only for example spinner in logic analyser its something like **spinner_channel_select_la1**

asitava1998, 2017-06-10:
I was thinking of adding a few points regarding the same in the README in the code practices & close the issue.

viveksb007, 2017-06-10:
Yeah we can add a point for this, but its more of a code style. 
We can create CodeStyle.md in docs or add a code-style section in README.md itself. Create 2-3 points under code-style section in README.md for now.
We will move it under CodeStyle.md in docs when README.md is quite long.

commit 58e24414 (2017-06-11): Fixes #144 Added Code style in README (#153)

* Fixes #144 Added Code style in README

* #Fixes 144 Added Code style in README
closed 2017-06-11 by viveksb007