fossasia/pslab-app #152: Add Tablet screen size support to Oscilloscope UI

akarshan96, 2017-06-10:
Make current Oscilloscope UI to support tablet screen size ( 7''  and 10''). Planning to increase text sizes, spaces between widgets and margins.

[image]
Screenshot of Oscilloscope Activity running in 7'' tablet.

akarshan96, 2017-06-11:
I have made some changes in Oscilloscope UI for tablets. The following is the screenshot for the same. 
@jithinbp @viveksb007 @asitava1998 @CloudyPadmal please suggest improvements.

[image]'
[image]
[image]

viveksb007, 2017-06-11:
@akarshan96 there is a lot of empty space in between. Could you try to increase graph size that would decrease panel size and you can also bring all those spinners in single line for large screens.

akarshan96, 2017-06-11:
@viveksb007 Yeah, on it. 👍

CloudyPadmal, 2017-06-11:
Do we need to add a round corners in the white panel? :thinking:

asitava1998, 2017-06-11:
@akarshan96  We can take advantage of bigger screens to allocate more space to the graph. The side bar looks really large, we need to think of something since reducing its size means it won't cover the entire width. The font sizes can be increased slightly.

akarshan96, 2017-06-11:
@asitava1998 I have reduced the size of both docks. Now the graph size is 3/4 of the height of the tablet and 7/8 of the with of the tablet (acording to the landscape mode). Let me know if they are required to be deacresed more. @CloudyPadmal I have remove boders from the lower dock, what you should I remove them?

[image]
**Screenshot for Nexus 7'' tablet**

[image]

 **Screenshot for Nexus 10'' tablet**

closed 2017-06-13 by akarshan96