fossasia/pslab-app #156: Supporting API levels < 18  by using an alternative to UsbDeviceConnection.bulkTransfer

jithinbp, 2017-06-12:
bulkTransfer method used for device communications currently requires a minimum API level of 18.
For API levels <18, another method that uses write function on the device after checking the API version may be added.
The present code crashes on an API 15 device : Simmtronics XPAD 1010 10" tablet with the following runtime error:
```
06-11 19:16:43.584 2168-2168/org.fossasia.pslab W/dalvikvm: threadid=1: thread exiting with uncaught exception (group=0x40a121f8)
06-11 19:16:43.594 2168-2168/org.fossasia.pslab E/AndroidRuntime: FATAL EXCEPTION: main
                                                                  java.lang.NoSuchMethodError: android.hardware.usb.UsbDeviceConnection.bulkTransfer
                                                                      at org.fossasia.pslab.communication.CommunicationHandler.write(CommunicationHandler.java:149)
                                                                      at org.fossasia.pslab.communication.PacketHandler.sendByte(PacketHandler.java:60)
                                                                      at org.fossasia.pslab.communication.PacketHandler.getVersion(PacketHandler.java:43)
                                                                      at org.fossasia.pslab.communication.ScienceLab.getVersion(ScienceLab.java:431)
                                                                      at org.fossasia.pslab.activity.MainActivity.getHomeFragment(MainActivity.java:147)
                                                                      at org.fossasia.pslab.activity.MainActivity.access$000(MainActivity.java:46)
                                                                      at org.fossasia.pslab.activity.MainActivity$1.run(MainActivity.java:118)
                                                                      at android.os.Handler.handleCallback(Handler.java:605)
```

viveksb007, 2017-06-14:
@jithinbp could you please check [this apk](https://drive.google.com/file/d/0B30IF3fyX4Q3cUloa0l4UDFlODA/view?usp=drivesdk) with that tablet with API level 15?
I have modified the write function for devices with API level < 18

jithinbp, 2017-06-14:
@viveksb007 :+1: it says 'Device Connected Successfully' . No crashes. But I can't find any mention of the version number read back. I'll check with adb logcat and get back

jithinbp, 2017-06-15:
@viveksb007 
I believe there exists a permissions issue because the app successfully reads the version string sometimes.
How to reproduce this : 
 - Connect the PSLab, and the OS (Kitkat 4.0.4 APi 15) automatically opens a dialog asking if the PSLab app should be opened. In this case, the version is detected correctly . no problem.  Error log 1
- Launch the app manually, and it promptly crashes. Error log 2
- I could not reproduce the behaviour where the app launches successfully but the version string is not displayed.

<details><summary>Error log 1 : app auto-launched. works fine</summary><p>

```
06-15 12:29:54.110 1187-1191/org.fossasia.pslab I/dalvikvm: threadid=3: reacting to signal 3
06-15 12:29:54.130 1187-1191/org.fossasia.pslab I/dalvikvm: Wrote stack traces to '/data/anr/traces.txt'
06-15 12:29:54.130 1187-1187/org.fossasia.pslab D/CommunicationHandler: VID : 3034PID : 33142
06-15 12:29:54.140 1187-1187/org.fossasia.pslab D/CommunicationHandler: VID : 1240PID : 223
06-15 12:29:54.140 1187-1187/org.fossasia.pslab D/CommunicationHandler: Found PSLAB Device
06-15 12:29:54.140 1187-1187/org.fossasia.pslab D/CommunicationHandler: Claiming interfaces, count=3
06-15 12:29:54.140 1187-1187/org.fossasia.pslab D/CommunicationHandler: Control interface=UsbInterface[mId=0,mClass=2,mSubclass=2,mProtocol=1,mEndpoints=[Landroid.os.Parcelable;@4105dc80]
06-15 12:29:54.140 1187-1187/org.fossasia.pslab D/CommunicationHandler: Control endpoint direction: 128
06-15 12:29:54.140 1187-1187/org.fossasia.pslab D/CommunicationHandler: Claiming data interface.
06-15 12:29:54.150 1187-1187/org.fossasia.pslab D/CommunicationHandler: data interface=UsbInterface[mId=1,mClass=10,mSubclass=0,mProtocol=0,mEndpoints=[Landroid.os.Parcelable;@4108e120]
06-15 12:29:54.150 1187-1187/org.fossasia.pslab D/CommunicationHandler: Read endpoint direction: 128
06-15 12:29:54.150 1187-1187/org.fossasia.pslab D/CommunicationHandler: Write endpoint direction: 0
06-15 12:29:54.350 1187-1187/org.fossasia.pslab D/MainActivity: PSLab device found
06-15 12:29:54.350 1187-1187/org.fossasia.pslab I/dalvikvm: Could not find method android.widget.ImageButton.hasOverlappingRendering, referenced from method android.support.v7.widget.AppCompatImageButton.hasOverlappingRendering
06-15 12:29:54.350 1187-1187/org.fossasia.pslab W/dalvikvm: VFY: unable to resolve virtual method 23317: Landroid/widget/ImageButton;.hasOverlappingRendering ()Z
06-15 12:29:54.350 1187-1187/org.fossasia.pslab D/dalvikvm: VFY: replacing opcode 0x6f at 0x0008
06-15 12:29:54.380 1187-1187/org.fossasia.pslab E/dalvikvm: Could not find class 'android.graphics.drawable.RippleDrawable', referenced from method android.support.v7.widget.AppCompatImageHelper.hasOverlappingRendering
06-15 12:29:54.380 1187-1187/org.fossasia.pslab W/dalvikvm: VFY: unable to resolve instanceof 160 (Landroid/graphics/drawable/RippleDrawable;) in Landroid/support/v7/widget/AppCompatImageHelper;
06-15 12:29:54.380 1187-1187/org.fossasia.pslab D/dalvikvm: VFY: replacing opcode 0x20 at 0x000c
06-15 12:29:54.410 1187-1187/org.fossasia.pslab D/ViewRootImpl: pckname = org.fossasia.pslab
06-15 12:29:54.420 1187-1187/org.fossasia.pslab V/CommunicationHandler: Read something[B@41164cf8
06-15 12:29:54.420 1187-1187/org.fossasia.pslab V/CommunicationHandler: Read something[B@41164cf8
06-15 12:29:54.610 1187-1191/org.fossasia.pslab I/dalvikvm: threadid=3: reacting to signal 3
06-15 12:29:54.620 1187-1191/org.fossasia.pslab I/dalvikvm: Wrote stack traces to '/data/anr/traces.txt'
06-15 12:29:54.920 1187-1187/org.fossasia.pslab E/CommunicationHandler: Read Error: 14
06-15 12:29:54.940 1187-1187/org.fossasia.pslab D/libEGL: loaded /system/lib/egl/libEGL_mali.so
06-15 12:29:54.950 1187-1187/org.fossasia.pslab D/libEGL: loaded /system/lib/egl/libGLESv1_CM_mali.so
06-15 12:29:54.960 1187-1187/org.fossasia.pslab D/libEGL: loaded /system/lib/egl/libGLESv2_mali.so
06-15 12:29:55.000 1187-1187/org.fossasia.pslab D/OpenGLRenderer: Enabling debug mode 0
06-15 12:29:55.010 1187-1187/org.fossasia.pslab I/dalvikvm: Could not find method android.support.v7.widget.LinearLayoutCompat.drawableHotspotChanged, referenced from method android.support.design.internal.ForegroundLinearLayout.drawableHotspotChanged
06-15 12:29:55.020 1187-1187/org.fossasia.pslab W/dalvikvm: VFY: unable to resolve virtual method 19281: Landroid/support/v7/widget/LinearLayoutCompat;.drawableHotspotChanged (FF)V
06-15 12:29:55.020 1187-1187/org.fossasia.pslab D/dalvikvm: VFY: replacing opcode 0x6f at 0x0000
06-15 12:29:55.040 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
06-15 12:29:55.040 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
06-15 12:29:55.050 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
06-15 12:29:55.050 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
06-15 12:29:55.050 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
06-15 12:29:55.050 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
06-15 12:29:55.050 1187-1187/org.fossasia.pslab W/ResourceType: Skipping entry 0x109003a in package table 0 because it is not complex!
[... cut ...]
06-15 12:33:59.850 3203-3206/org.fossasia.pslab I/dalvikvm: threadid=3: reacting to signal 3
06-15 12:33:59.850 3203-3206/org.fossasia.pslab I/dalvikvm: Wrote stack traces to '/data/anr/traces.txt'
06-15 12:34:00.050 3203-3206/org.fossasia.pslab I/dalvikvm: threadid=3: reacting to signal 3
06-15 12:34:00.050 3203-3206/org.fossasia.pslab I/dalvikvm: Wrote stack traces to '/data/anr/traces.txt'
06-15 12:34:01.170 3203-3203/org.fossasia.pslab I/Process: Sending signal. PID: 3203 SIG: 9

```
</p></details>

viveksb007, 2017-06-15:
Yeah I got the error, the app doesn't have functionality to obtain permission explicitly. I will add it and give link of new apk for testing. Thanks for the logs.

PR #186 (2017-06-17): modified write function to support device with API level < 18
Partly Fixes #156 

Changes: modified write function to support device with API level < 18

Tested by jithin on device with API level 15, there was  a permission issue #179  which @akarshan96 is taking care of.
closed 2017-06-17 by mariobehling