fossasia/pslab-app #170: High Light Tab strip in Control Activity

viveksb007, 2017-06-14:
**Actual Behaviour**

Tab-Strip doesn't indicate which tab is opened currently.

**Expected Behaviour**

Tab-strip should high-light the section of tab-strip showing which tab/fragment is opened currently.

**Steps to reproduce it**

Open Control Activity and switch between tabs, tab won't indicate which one is opened currently.

**LogCat for the issue**

None

**Screenshots of the issue**

[image]

@asitava1998  Have a look at it. Please!

asitava1998, 2017-06-14:
@viveksb007  I think it would be better if the active tab is slightly bulged from background or maybe the text colour is changed from white to any other colour. I will work on it but let us just fix how we want to represent a highlighted tab. What do you think the representation should be done

viveksb007, 2017-06-14:
Changing color of text and image from white -> black,etc should be okay to show the high-light. Something like lower tab-strip of Paytm.

commit 2389f520 (2017-06-19): Fixes #170 highlights bottomnavigation bar (#196)
closed 2017-06-19 by viveksb007