fossasia/pslab-app #175: App crashes when runInitSequence method from ScienceLab.java is called

akarshan96, 2017-06-15:
**Actual Behaviour**

Calling runInitSequence method from ScienceLab.java is causing the app to crash.

**Expected Behaviour**

State here what the feature should enable the user to do.

**Steps to reproduce it**

Create an object of ScienceLab and call runInitSequence method.

**LogCat for the issue**

```
06-15 17:30:35.206 18841-18841/org.fossasia.pslab E/AndroidRuntime: FATAL EXCEPTION: main
                                                                    Process: org.fossasia.pslab, PID: 18841
                                                                    java.lang.RuntimeException: Unable to start activity ComponentInfo{org.fossasia.pslab/org.fossasia.pslab.activity.OscilloscopeActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.fossasia.pslab.communication.PacketHandler.isConnected()' on a null object reference
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2423)
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483)
                                                                        at android.app.ActivityThread.access$900(ActivityThread.java:153)
                                                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                        at android.os.Looper.loop(Looper.java:148)
                                                                        at android.app.ActivityThread.main(ActivityThread.java:5438)
                                                                        at java.lang.reflect.Method.invoke(Native Method)
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738)
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628)
                                                                     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.fossasia.pslab.communication.PacketHandler.isConnected()' on a null object reference
                                                                        at org.fossasia.pslab.communication.peripherals.NRF24L01.<init>(NRF24L01.java:92)
                                                                        at org.fossasia.pslab.communication.ScienceLab.runInitSequence(ScienceLab.java:147)
                                                                        at org.fossasia.pslab.activity.OscilloscopeActivity.onCreate(OscilloscopeActivity.java:144)
                                                                        at android.app.Activity.performCreate(Activity.java:6303)
                                                                        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1108)
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2376)
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483) 
                                                                        at android.app.ActivityThread.access$900(ActivityThread.java:153) 
                                                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349) 
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                        at android.os.Looper.loop(Looper.java:148) 
                                                                        at android.app.ActivityThread.main(ActivityThread.java:5438) 
                                                                        at java.lang.reflect.Method.invoke(Native Method) 
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738) 
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628) 
```

**Screenshots of the issue**

none

**Would you like to work on the issue?**

Yes

viveksb007, 2017-06-15:
@akarshan96 could you also push your code changes to some branch of your repo. Would help in understanding the issue better.

akarshan96, 2017-06-15:
@viveksb007, the issue is solved. `y.clear()` [here](https://github.com/fossasia/pslab-android/blob/master/app/src/main/java/org/fossasia/pslab/communication/ScienceLab.java#L2584) is also clearing the `points` ArrayList. Let me run some more tests. 
Thanks

akarshan96, 2017-06-16:
Fixed in #180

closed 2017-06-16 by akarshan96