fossasia/pslab-app #176: control_main_list_item has a few issues in the view

CloudyPadmal, 2017-06-15:
**Actual Behaviour**

The `control_main_list_item.xml` ;
- Doesn't contain padding on left and right
- Text boxes allow letters other than numerical values
- Text boxes don't have a padding
- `SET` button is having default colors

**Expected Behaviour**

- Cardview needs a padding on the sides
- Text boxes can be allowed to have numerical values to avoid programming errors occurring in future when functionalities are to be implemented
- Text boxes should have a padding to display content not sticked to the border
- `SET` button should follow the primary color as rest of the buttons

**Steps to reproduce it**

- Open app
- Go to `Applications`
- Click on `Control` and go to main control section and observe

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes!

asitava1998, 2017-06-15:
@CloudyPadmal Before you start, let me have a look at it. The view on this screen is distorted. What screen size are you using?

CloudyPadmal, 2017-06-15:
Yes @asitava1998 there is a text box on the right side but it's not having enough space to be displayed. My screen is `480 x 800`!

asitava1998, 2017-06-15:
@CloudyPadmal Can you create another issue with this screenshot and mention the details of the problem ? Then I can work on the screen compatibility.

CloudyPadmal, 2017-06-15:
Sure @asitava1998 ! :+1:

commit 2632b070 (2017-06-16): Fix #176 : Main Control Panel UI Fixes (#181)
closed 2017-06-16 by wavicles