fossasia/pslab-app #179: Add code to obtain permission to access USB device explicitly

viveksb007, 2017-06-15:
**Actual Behaviour**

Currently App crashes when opened manually launched as App doesn't have permission to access USB device explicitly. Previously Activity is launched by ```intent``` ```action-filter``` and permission is given by system.

**Expected Behaviour**

If permission is not given by system, then ask user to access USB device, this would prevent app from crashing.

**Steps to reproduce it**

Latest device have option to check, and them system gives permission automatically in manual case too. But in devices with API level 15, system might not be able to give permission automatically.

**LogCat for the issue**

Log is available in Parent Issue : https://github.com/fossasia/pslab-android/issues/156

**Screenshots of the issue**

None

**Would you like to work on the issue?**

Yes

viveksb007, 2017-06-16:
@akarshan96 I would make a PR of modified write function to support API level < 18.
You can add code for obtaining USB permission explicitly.
Thanks

akarshan96, 2017-06-19:
@viveksb007 @asitava1998 @CloudyPadmal, here is the [link](https://drive.google.com/file/d/0B423XEmWL9vKTkJ0SFgzb0YwVmM/view?usp=sharing) to new apk. I am facing an issue. There are 2 permission asked in this apk
- To open PSLab when the USB device connected to the phone.
- Allow PSLab to access USB accessory.

The second one ie obtain permission to access USB device explicitly is recently added by me. I have referred [this](https://developer.android.com/guide/topics/connectivity/usb/host.html) source for the same. (I have moved this permission to the splash activity) 

screenshots of the second permission

[image]

If I cancel.....

[image]

For the first permission if I hit cancel, error arises

screenshots of the first permission

[image]

and if I hit cancel and open the app again, app crashes with the logs below

```
6-19 16:47:19.790 30242-30242/org.fossasia.pslab E/AndroidRuntime: FATAL EXCEPTION: main
                                                                    Process: org.fossasia.pslab, PID: 30242
                                                                    java.lang.RuntimeException: Unable to start activity ComponentInfo{org.fossasia.pslab/org.fossasia.pslab.activity.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.hardware.usb.UsbDeviceConnection.claimInterface(android.hardware.usb.UsbInterface, boolean)' on a null object reference
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2423)
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483)
                                                                        at android.app.ActivityThread.access$900(ActivityThread.java:153)
                                                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                        at android.os.Looper.loop(Looper.java:148)
                                                                        at android.app.ActivityThread.main(ActivityThread.java:5438)
                                                                        at java.lang.reflect.Method.invoke(Native Method)
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738)
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628)
                                                                     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.hardware.usb.UsbDeviceConnection.claimInterface(android.hardware.usb.UsbInterface, boolean)' on a null object reference
                                                                        at org.fossasia.pslab.communication.CommunicationHandler.open(CommunicationHandler.java:81)
                                                                        at org.fossasia.pslab.communication.ScienceLab.<init>(ScienceLab.java:85)
                                                                        at org.fossasia.pslab.others.ScienceLabCommon.openDevice(ScienceLabCommon.java:23)
                                                                        at org.fossasia.pslab.activity.MainActivity.onCreate(MainActivity.java:94)
                                                                        at android.app.Activity.performCreate(Activity.java:6303)
                                                                        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1108)
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2376)
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2483) 
                                                                        at android.app.ActivityThread.access$900(ActivityThread.java:153) 
                                                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349) 
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                        at android.os.Looper.loop(Looper.java:148) 
                                                                        at android.app.ActivityThread.main(ActivityThread.java:5438) 
                                                                        at java.lang.reflect.Method.invoke(Native Method) 
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738) 
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628) 

```

viveksb007, 2017-06-19:
@akarshan96 Don't move permission to splash Activity. Check in MainActivity if its started by system intent or by user clicking on App. Usinh getIntent.getAction Google it you will know. According to this you need to choose if user gave permission when system dialog popped you don't need need to ask permission explicitly else ask user to access usb device explicitly.

viveksb007, 2017-06-19:
Yeah splash Activity is started first and initialisation isn't shifted currently. So when usb device object would be created in splash Activity, it can't be passed across Activity i.e to Main Activity for initialisation. That's why I was saying to obtain permission in MainActivity itself. If obtaining permission and initialisation both are handled in splash Activity then its okay, but that's out of scope for this issue.

viveksb007, 2017-06-19:
@akarshan96 If the issue still persists, I will have a look at it. You can start with implementing Oscilloscope Functionality and open relevant issues. Just push the work you did yet to some branch of your repo and provide link here.
Thanks

akarshan96, 2017-06-19:
@viveksb007 sure, I was thinking to do this part once we move initialization to Splash Activity. Here is a [link](https://github.com/akarshan96/pslab-android/blob/permissions/app/src/main/java/org/fossasia/pslab/activity/MainActivity.java). 
Thanks :)

viveksb007, 2017-06-19:
@akarshan96 @jithinbp @asitava1998 here is the link of apk for explicit permission [apk link](https://drive.google.com/open?id=0B30IF3fyX4Q3azJXemVfRmpvTlU). Please try with devices denying and opening app in any case and if there is a crash. Please report with the logs.
Thanks.

jithinbp, 2017-06-19:
Awesome . No crashes.
I've tried the following
- connected the hardware. Accepted the auto-launch dialog.
- connected the hardware. Cancelled the auto-launch dialog. Opened the app manually. Granted permission.

- opened the app manually. Denied permission. ( In this case we should probably show some message, or a button to request permission again )

Device Disconnect events should also be handled. Currently the app doesn't know if the hardware is removed, and this can cause crashes further down the line

Cheers

akarshan96, 2017-06-19:
It works fine 👍 
Thanks

asitava1998, 2017-06-19:
Works fine. No crashes. Good work :+1:

viveksb007, 2017-06-20:
Thanks everyone.
@jithinbp there is already an issue https://github.com/fossasia/pslab-android/issues/72 about handling abrupt disconnection. I would ask about improvement and address the issue.

viveksb007, 2017-06-21:
Fixed in #206

closed 2017-06-21 by viveksb007