fossasia/pslab-app #182: Main Control UI seekbars and buttons not working

CloudyPadmal, 2017-06-16:
**Actual Behaviour**

Since this is still at the UI designing level, functionalities related to user interactions haven't been implemented yet. Hence the increment, decrement buttons not responding and seek bars not updating accordingly

**Expected Behaviour**

- When user clicks on `+` button, value in the text box should increase by one unit
- When user clicks on `-` button, value in the text box should decrease by one unit
- Seek bar and the value in text box should be inter-related
- There should be limits to the values a user can enter into the text boxes

**Steps to reproduce it**

- Open app
- Go to `Applications`
- Open `Control` and go to main control section
- Enter any value to text boxes and try clicking buttons and observe

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

I would like to work on this! :+1:

asitava1998, 2017-06-16:
Well the problem is because I am yet to complete issue #148 which is responsible for this and along with synchronising the movement of seekbar with the edittext.

viveksb007, 2017-06-16:
@CloudyPadmal , first let him solve issue related to UI and implementation of functionality. Issue #148 is almost same with less details.

CloudyPadmal, 2017-06-16:
Ah now it's cleared! :+1: Let's do like that! @asitava1998 @viveksb007 :)

viveksb007, 2017-07-09:
Fixed in #189

closed 2017-07-09 by viveksb007