fossasia/pslab-app #193: Handling faulty hardware attached

jithinbp, 2017-06-18:
**Actual Behaviour**

If for some reason the processor in the PSLab hardware does not respond, the app freezes for over a minute or two instead of detecting a fault in the hardware 

A few minutes later it says 'initialization complete' without actually realising the error

**Expected Behaviour**
report a hardware error. Do not try to load calibration.

**Steps to reproduce it**
Reprogram the PSLab with some firmware that does not implement the PSLab state machine

**LogCat for the issue**
```
D/dalvikvm(22783): VFY: replacing opcode 0x6f at 0x0008
D/dalvikvm(22783): GC_FOR_ALLOC freed 119K, 3% free 7698K/7879K, paused 41ms
I/dalvikvm-heap(22783): Grow heap (frag case) to 8.547MB for 1048592-byte allocation
D/dalvikvm(22783): GC_CONCURRENT freed 10K, 3% free 8711K/8967K, paused 2ms+2ms
D/dalvikvm(22783): GC_FOR_ALLOC freed 2K, 3% free 8718K/8967K, paused 17ms
I/dalvikvm-heap(22783): Grow heap (frag case) to 9.543MB for 1048592-byte allocation
D/dalvikvm(22783): GC_FOR_ALLOC freed <1K, 4% free 9741K/10055K, paused 18ms
I/ActivityManager(  158): Displayed org.fossasia.pslab/.activity.MainActivity: +1s534ms
E/CommunicationHandler(22783): Read Error: 0
D/dalvikvm(22783): GC_FOR_ALLOC freed 1071K, 12% free 9610K/10887K, paused 19ms
D/dalvikvm(22783): GC_CONCURRENT freed 2K, 3% free 10624K/10887K, paused 1ms+2ms
E/CommunicationHandler(22783): Read Error: 0
E/CommunicationHandler(22783): Read Error: 0
E/CommunicationHandler(22783): Read Error: 0
D/audio_hw_primary(   90): start_output_stream: card:0, port:0, rate:44100
D/OpenGLRenderer(  223): Flushing caches (mode 0)
E/CommunicationHandler(22783): Read Error: 0
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in reading byte
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
E/NRF24L01(22783): Radio transceiver not installed/not found
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
V/ScienceLab(22783): Cap and PCS calibration invalid
I/Process (  158): Sending signal. PID: 22783 SIG: 3
I/dalvikvm(22783): threadid=3: reacting to signal 3
I/dalvikvm(22783): Wrote stack traces to '/data/anr/traces.txt'
D/OpenGLRenderer(  223): Flushing caches (mode 1)
W/ActivityManager(  158): Activity idle timeout for ActivityRecord{41072f90 org.fossasia.pslab/.activity.MainActivity}
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
```
The timeouts go on for a while. Interspersed with a message from the OS regarding preparation of a crash report
```
I/PlayCommon(21207): [546] com.google.android.play.a.g.e(220): Preparing logs for uploading
I/PlayCommon(21207): [546] com.google.android.play.a.g.e(224): No file ready to send
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading
E/CommunicationHandler(22783): Read Error: 0
E/PacketHandler(22783): Error in packetHandler Reading

and so on for a long while...

^C

```
**Screenshots of the issue**
start screen stuck at 'initializing ..' 

**Would you like to work on the issue?**
nope. I can help with testing newer versions.

hardik124, 2017-08-09:
@viveksb007  can you elaborate on this a little more ?

viveksb007, 2017-08-10:
Sure,
Most  of it is covered in issue description, if some faulty device is connected, then app should be capable of telling that the device connected is not functioning as intended and provide user with appropriate message/instructions.

isaeef, 2017-09-11:
Is this issue still open , I want to fix this issue.

viveksb007, 2017-09-11:
@isaeef yeah its still open. Go ahead 👍

batbrain7, 2017-10-18:
@viveksb007 since there is no work on the issue, may I fix it?

viveksb007, 2017-10-18:
@batbrain7 sure go ahead 👍

CloudyPadmal, 2018-07-16:
Current firmware does not go into state machine unless it is fully initiated.

closed 2018-07-16 by CloudyPadmal