fossasia/pslab-app #200: Implement captureOne method in Oscilloscope UI.

akarshan96, 2017-06-19:
**Actual Behaviour**

Since we are done with basic Oscilloscope UI, functionality needs to be added to UI. Currently, Oscilloscope UI is not functional.

**Expected Behaviour**

Fetch an oscilloscope trace from the specified input channel and plot it into the graph.

**Steps to reproduce it**

Open Oscilloscope UI and under Channel Parameters select any one input channel.

**LogCat for the issue**

none

**Screenshots of the issue**

none.

**Would you like to work on the issue?**

Yes

jithinbp, 2017-06-20:
@akarshan96 I wouldn't recommend the captureOne routine for a continuously
running oscilloscope UI. capture1 is a blocking call, and since the
communication routines are not in a separate thread for now, it will make
for a super laggy UI.

Instead, you can look at the source of capture1 which has 3 parts
- sending the acquisition parameters to the hardware
- time.sleep for the acquisition duration
- fetch the acquired data.
You can simply use 1,3 so that the UI can be kept responsive during the
waiting period

akarshan96, 2017-06-21:
@jithinbp, here is the video [link](https://goo.gl/photos/juS9ngFqgpgo21LK9). I have called communication routines in the background thread as you said.

viveksb007, 2017-06-21:
@akarshan96 Looks great.
Is that real-time too or updating after some specific defined interval?
And are there any packet drops or error in reading values from device?

akarshan96, 2017-06-21:
@viveksb007 thanks. It's real-time. Yes, there are some read errors.

jithinbp, 2017-06-21:
@akarshan96 <https://github.com/akarshan96> thanks :-)
 Looks good, but it feels slower than what I had expected.

   - What is the number of points being recorded? We can record up to 10K
   samples in a single capture, but that is major overkill for an oscilloscope
   where the graph will have only around 600 pixels along the x-axis.
   We can record 500 points per channel, so that the refresh rate is much
   higher.

   - is there any extra delay between acquisitions? Since it's a separate
   thread, we can completely do away with it, I suppose.

   - The graph's scaling seems to be off because the microphone channel's
   signal is outside the +/-3.3V range. You could check that out also
   - The retrieve_buffer call is broken up into sections. What is the value
   of the data_splitting variable? Try to see if you can increase it without
   getting packet drops (Don't use a very high end phone if possible. I'll
   test it on my low-end tablet as well).

Thanks

akarshan96, 2017-06-22:
@jithinbp, I have recorded 1000 points and kept the time gap of 10 microseconds. I will try recording 500 points and see the performance. I have skipped delay between the acquisitions. For 1000 samples, the value of data splitting variable is 200.

jithinbp, 2017-06-22:
@akarshan96  , 10mS of acquisition time is negligible, so it seems the maximum time is taken by data transfer. Could you please measure the following durations:
+ total time taken to receive 1000 data points 
+ time taken for each subsection of 200 points
thanks

viveksb007, 2017-06-22:
I think delay is due to read error, when I was reading bytes for initialisation, it took almost a minute.
Most probably error is due to getAcknowledgement as it tries to read a single byte and reading a single byte was giving read error. That's why readline method couldn't be implemented yet. So in readFlash method, I read 17 bytes directly instead of 16+1. That worked.
@jithinbp what if we try to read 16 bytes during getAcknowledgement and return byte at position 0.
As it reads 16 bytes quite well during reading flash data. 
@akarshan96 can you try this that if its gives error on reading 16 bytes during acknowledgment or reads garbage bytes successfully?

jithinbp, 2017-06-22:
@viveksb007 He's not reading the calibration data. He's reading 1000 points in sections of 200 each.
Which means, there is a possibility of the acknowledgement byte getting timed out 5+1 times. 
@akarshan96 Please measure the time taken to read 200 points (400 bytes), and time taken to read the ack byte. vivek is right, you should read 401 bytes, and only process the first 400.

viveksb007, 2017-06-22:
I was trying to say that at each iteration here https://github.com/fossasia/pslab-android/blob/master/app/src/main/java/org/fossasia/pslab/communication/ScienceLab.java#L879 , first we getAcknowledgement and then we tell device to send more bytes right?
So at getAcknowledgement call in loop buffer has only acknowledgement byte. So if we read more that 1 byte say 16 and return buffer[0]. If that works then we can make changes in getAcknowledgement method, so we don't have to change everywhere in code.
So the question is when device wrote a single byte serially and we try to read more bytes. Do we get read error or we read garbage bytes?

jithinbp, 2017-06-22:
@viveksb007 okay, i understand now.  but I believe trying to read 16 bytes where only one is expected will result in  a timeout error, and slow down things. I can propose the following:
Read the following data in a single chunk of 4 bytes
```
           conversionDone = mPacketHandler.getByte();
            samples = mPacketHandler.getInt();
            mPacketHandler.getAcknowledgement();
```
and then use the first three for `conversionDone` and `samples` . Will this work?

viveksb007, 2017-06-22:
Sorry for wrong link, I was pointing to [this](https://github.com/fossasia/pslab-android/blob/master/app/src/main/java/org/fossasia/pslab/communication/ScienceLab.java#L906) line. 
And yeah this should work. We just want to avoid reading single byte.
Will do some testing about how much bytes I can read reliably. And will share the findings.

jithinbp, 2017-06-22:
in line 904  `(int j = 0; j < data.length - 1; j++)` , won't this loop run 399 times instead of 400? 

you could combine the acknowledgement read with the data read
```
                mPacketHandler.sendInt(this.dataSplitting);
                mPacketHandler.sendInt(i * this.dataSplitting );
                byte[] data = new byte[this.dataSplitting * 2 +1];
                mPacketHandler.read(data, this.dataSplitting * 2 +1 ); //+1 for the ack byte
                for (int j = 0; j < data.length - 1; j++)  // copy first 400 bytes. assuming the split is 200 integers.
                    listData.add(data[j]);
                //mPacketHandler.getAcknowledgement(); //no longer required. data[this.dataSplitting * 2] = ack byte
```

viveksb007, 2017-06-22:
Yeah my bad, would open an issue and rectify that loop iteration range wherever its incorrect.
And incorporating ack byte with 400 bytes would work. But getAcknowledgement is used almost everywhere, if we could find some way of making it work from packetHandler itself. That would save a lot of modification.

akarshan96, 2017-06-22:
@jithinbp @viveksb007 It takes 6 - 9 millisec to send the acquisition parameters and 3 - 5 seconds to receive the data.

jithinbp, 2017-06-22:
@akarshan96 3-5 seconds sounds like some sort of timeout error. At a BAUD rate of 1e6, you're attempting to read 200*2*8= 3200 bits. which after accounting for overheads, should be less than a few ten mS, right? We could try to combine the acknowledge byte with the data bytes , and then reject it later, if that's where the timeout issue lies

PR #215 (2017-06-24): Initial steps for implementing CaptureOne method in Oscilloscope UI
Fixes #200 and #212 

Changes: 
- Implemented captureOne method
- Prevented Oscilloscope Activity to sleep while running.

Things still left to be done
- Remove read errors
- Determine no. of samples and time-gaps 
- speed up the process

Screenshots for the change: 

[image]
closed 2017-06-24 by viveksb007