fossasia/pslab-app #202: Add remaining functionalities of Control Main

asitava1998, 2017-06-20:
**Actual Behaviour**

* Two of the buttons increment and SET do not work.
* The seekbar does not synchronize with the data entered in edittext (although vice-versa is true).

**Expected Behaviour**

* All the buttons should be functional.
* The seekbar position should also change with the change of values in edit text.

**Steps to reproduce it**

Main menu -> Applications -> Control -> Control

**LogCat for the issue**

None

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

asitava1998, 2017-06-27:
Part of it has been fixed in issue #234 . The only part remaining is that of updating the seekbar when the numbers are entered in editText .

viveksb007, 2017-06-27:
I did that but that's what causes precision issue that i discussed yesterday. Updating seekbar would also trigger seekBarProgress method which takes integer progess and updates editText. Thats results in losing precision. You can see its loopback.

viveksb007, 2017-06-27:
What are textview after **Set** button for?
And I saw option to set W1 and W2 in advanced section. So what is wave1, wave2 for?

asitava1998, 2017-06-27:
Oh sorry, I forgot about that discussion on seekbar on the channel.
The textView was after `SET` was meant for displaying the results of `getPV1`, `getPV2` etc. 

I had created the designs after having a look at the desktop app. So, W1(wave1) & W2(wave2) are capable of generating both sine & square waves which is done from the Advanced section. In the Main section, W1 & W2 are for generating sine waves only and Square generates square wave using W1. This was the implementation in the desktop app. Even I feel this feature is sort of redundant and can be removed from main after a discussion on the channel. What say?

asitava1998, 2017-08-29:
@viveksb007 I think we can close this issue. The changes mentioned here have already been done through other issues and PRs.

viveksb007, 2017-08-29:
@asitava1998 Yeah Points mentioned in the Issues description has been achieved by other Issues and PRs.

closed 2017-08-29 by viveksb007