fossasia/pslab-app #203: Different Slider at Oscilloscope UI

CloudyPadmal, 2017-06-20:
**Actual Behaviour**

In `Channel Parameter`, the slider related to `Trigger` is different from the one implemented in the application

**Expected Behaviour**

All the sliders in the app should share similar characteristics. Hence this slider needs to be changed

**Steps to reproduce it**
1. Open app
2. Go to `Applications`
3. Select `Oscilloscope`
4. Click on `Timebase and Trigger` tab and observe

**LogCat for the issue**
> Not available

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes!

viveksb007, 2017-06-20:
@CloudyPadmal both sliders are same in my Android device. Api level 23. Whats the api level of your device? Could this be due to different Android versions?

CloudyPadmal, 2017-06-20:
@viveksb007 I checked this with an API level 17 device!

commit 21ed8c8f (2017-06-20): Fix #203 : Common Slider in Oscilloscope (#204)
closed 2017-06-20 by viveksb007