fossasia/pslab-app #211: add a script for uploading apk generated from PR to apk branch of repo.

viveksb007, 2017-06-21:
**Actual Behaviour**

Currently No apk is generated when a PR is made to repo. So apk has to be uploaded somewhere and link for that apk is given to fellow devs for testing. 

**Expected Behaviour**

Whenever a PR is made to repo corresponding apk should be made in apk branch and that apk should be deleted from apk branch when PR is merged. 

**Would you like to work on the issue?**

Yes

viveksb007, 2017-06-23:
@CloudyPadmal  I see [here](https://github.com/fossasia/phimpme-android/blob/development/upload_apk_forks.sh) you have put your API key. It should be private. And did it worked i.e was apk generated successfully whenever any PR is made?

CloudyPadmal, 2017-06-23:
@viveksb007 in the script whenever we made a PR, we were supposed to change the following lines in the script; :)
```
# *** Important: For everychange pls update the below line in the following format:
# git clone --quiet --branch=apk htpps://username:authentiToken@github.com/username/phimpme-android apk > /dev/null
# replacing username with your github username and authentiToken with a personal token for your account.***
# Your token is advised to have only access to public repositories. Its only needed.
```

viveksb007, 2017-08-28:
Not required as when many PRs are stacked apk would be replaced and dev who sends the PR attaches the apk in PR description.

closed 2017-08-28 by viveksb007