fossasia/pslab-app #216: Main thread and Initialisation thread routines are coinciding resulting in wrong version string

viveksb007, 2017-06-22:
**Actual Behaviour**

Main Thread (UI thread) and initialisation threads routines coincide when version read is done in between initialisation is going on and version string is set up as some part of calibration data.

**Expected Behaviour**

It shouldn't do a version read between initialisation.

**Steps to reproduce it**

Open App go to Application Fragment and switch back before initialisation is completed.

**LogCat for the issue**

None

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

viveksb007, 2017-06-23:
Fixed in #217

closed 2017-06-23 by viveksb007

jithinbp, 2017-06-23:
Shouldn't we have like a common thread for everything? One that is not linked to any activity.
I don't know how this is implemented in Java, but in Python I once wrote a communication thread with which data is exchanged via signals and corresponding slots.
E.g., If I want to read the buffer, I would pass the necessary parameters to the communication handler, and also designate a callback function to which the communication handler will pass the buffer when it has finished handling. My main UI then picks up execution from the callback function.
But this approach requires all calls to go through the communication thread in order to avoid clashes

viveksb007, 2017-06-23:
Yes I thought about it too. To make a thread which has same lifetime as of UI thread and intercommunicate with each other. Haven't found a way to implement it in Android yet. Asyn task that @akarshan96 used is meant for small work max 4-5 sec. 
```
AsyncTask is designed to be a helper class around Thread and Handler and does not constitute a generic threading framework. AsyncTasks should ideally be used for short operations (a few seconds at the most.) If you need to keep threads running for long periods of time, it is highly recommended you use the various APIs provided by the java.util.concurrent package such as Executor, ThreadPoolExecutor and FutureTask.
```
For now we are creating a thread for single task and it gets killed after doing its task.
I would look for how to implement such architecture like its suggested to use ThreadPoolExecuter, FutureTask, etc. The clash happened in this case UI thread called for Version String and clashed with Initialisation thread. But I solved it here by Saving Version string in between when device is first connected and disconnected, So it doesn't call for version string again and again.