fossasia/pslab-app #225: Implement Logic Analyzer Functionality

viveksb007, 2017-06-24:
**Actual Behaviour**

Currently Logic Analyzer is generating randon digital signal.

**Expected Behaviour**

It should generate signal corresponding to selected channel mode.

**Steps to reproduce it**

NA

**LogCat for the issue**

NA

**Screenshots of the issue**

NA

**Would you like to work on the issue?**

Yes

viveksb007, 2017-07-12:
@CloudyPadmal If you don't have any issue at hand currently. Could you take this one up?
I did some work commits are on [dev](https://github.com/viveksb007/pslab-android/tree/dev) branch.

CloudyPadmal, 2017-07-13:
Sure @viveksb007 ! I'll give this a try! :+1: Thanks a lot!

asitava1998, 2017-08-29:
@CloudyPadmal  What is the current status of this issue?

CloudyPadmal, 2017-08-29:
Hi @asitava1998 ! Implementation is complete! But errors in byte reading cause issues. That's where it is stuck!

CloudyPadmal, 2018-05-28:
Will be addressed in #912

closed 2018-05-28 by CloudyPadmal