fossasia/pslab-app #228: Field values have a higher resolution than required in Control UI

CloudyPadmal, 2017-06-24:
**Actual Behaviour**

The Control UI input boxes has nearly 8 decimal places.

**Expected Behaviour**

The precision for voltages should be set to 3 decimal places & frequency should be set to one decimal place

**Steps to reproduce it**

1. Open App
2. Go to `Applications`
3. Proceed to `Control`
4. Try sliding the sliders to set values to input boxes and observe

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes!

asitava1998, 2017-06-24:
@CloudyPadmal I would suggest you to make a new layout directory `layout-sw320dp` and adjust the layouts for control in that file. Are you also going to take care about the digits after decimal or should I do it?

CloudyPadmal, 2017-06-25:
@asitava1998 I am planning on creating a new directory to support 320dp screens and adjust view changes there :) I'm working on this issue and will send the pull request soon! :+1:

commit 2ea6e727 (2017-06-27): Fixes #228 : Resolutions set to required amounts in control UI (#232)

* Fixes #228 : Resolutions set to required amounts in control UI

* Fixing Travis build failure

* Fix Travis build fail
closed 2017-06-27 by mariobehling