fossasia/pslab-app #247: Close Oscilloscope Activity when device is disconnected

akarshan96, 2017-06-28:
**Actual Behaviour**

Oscilloscope Activity doesn't close when USB is disconnected. 

**Expected Behaviour**

Oscilloscope Activity should close when USB is disconnected.

**Steps to reproduce it**

Menu Drawer -> Applications -> Oscilloscope -> detach PSLab device

**LogCat for the issue**

> None

**Screenshots of the issue**

> None

**Would you like to work on the issue?**

Yes

akarshan96, 2017-07-09:
@jithinbp @viveksb007 @asitava1998 @CloudyPadmal, here is the [video link](https://goo.gl/photos/TNCEJk1xiiCrCxAWA). I have added a LED icon which goes from red to green when data is read. Please provide your reviews

viveksb007, 2017-07-09:
Looks great @akarshan96.

PR #293 (2017-07-10): Added LED icon in Oscilloscope UI
Fixes #247 
Changes: 
- Added LED icon which goes from red to green when data is read.
- Reformatted code

Apk:
[app-debug.apk.zip](https://github.com/fossasia/pslab-android/files/1133964/app-debug.apk.zip)

Screenshots for the change: 

[image]

[image]
commit 1e19f345 (2017-07-10): Added LED icon in Oscilloscope UI (#293)

* Adds LED icon in UI

* reformats the code
Fixes #247
closed 2017-07-10 by viveksb007