fossasia/pslab-app #249: Read error during fetchTrace

viveksb007, 2017-06-29:
**Actual Behaviour**

There is read error during fetchTrace. Most probably due to timeout. Due to which chart update on oscilloscope Activity is quite slow.

**Expected Behaviour**

There should be any read error and oscilloscope Activity shouldn't be laggy.

**Steps to reproduce it**

Open oscilloscope Activity and see chart update is quite slow.

**Would you like to work on the issue?**

Yes

viveksb007, 2017-06-30:
@akarshan96 @jithinbp There are still read errors. But sometimes I can read complete 1000 sample data i.e 2000 bytes successfully.
[Screencast Link](https://goo.gl/photos/jVM2RSy2CDXx6idJ6)
[APK Link](https://drive.google.com/open?id=0B30IF3fyX4Q3anRwNEJIQldvYXc). @asitava1998 @CloudyPadmal you can also check the apk. Just connect W1 to CH1 before connecting device to Mobile and you should see sine wave. 
Should be move ahead from here?  because root cause for packet drop hasn't been found yet. ( Most probably buffer issue.) Its not Thread related, I checked for single cycle of CaptureTrace and fetchTrace.

akarshan96, 2017-06-30:
@viveksb007, it's faster now👍 . I am sharing the screenshot.

[image]

jithinbp, 2017-07-01:
Cool :+1:
But the amplitude seems to be +-40V . So either the scaling polynomial isn't right, or there is some issue with the graph. do check please. You can simply log max(ydaya), min(ydata) to know which.
Thanks

jithinbp, 2017-07-01:
Oh wait. You have set the range to +/-160V, and there is no 10Mohm resistor in series probably, so the amplitude must be around 3.3 V . Sorry.

If you look at the schematic, the input stage is an op-amp in the inverting configuration with RF=200K and Ri = 1M , so gain is 1/5 . When you add a series resistor of 10M , gain becomes 200K/(1M+10M)  , so the slopes of your calibration polynomials must be further divided by a factor of 11 for this  special case

viveksb007, 2017-07-01:
@jithinbp During unpacking bytes in fetchChannel, I assumed I am getting MSB first then LSB, could you confirm if that's the case?

jithinbp, 2017-07-01:
No. First LSB, then MSB.
Refer to getInt in this library. 
https://github.com/fossasia/pslab-firmware/blob/master/PSLab_Original/PSLAB_UART.c

viveksb007, 2017-07-01:
Okay I would change and test it. And also in sequence of bytes when some are dropped then from there unpacking for ```int``` would be wrong. As byte of some other ```int``` would be unpacked with byte of some other ```int```.

viveksb007, 2017-07-01:
@jithinbp Now after correcting bit unpacking I am getting this.
[image]

I am doing setSine1(3000)
followed by captureTraces(1, 1000, 10, "CH1", trigger = false, CH123SA = 0)
delay
fetchTrace(1)
What do you think is problem here?

jithinbp, 2017-07-01:
I think you are working in the wrong voltage range.  
check the setGain function. The programmable gain amplifiers are probably set to +/- 1V range or lower. Signal seems to be getting clipped
Troubleshooting steps
* Check the sine wave output with a separate oscilloscope if you have one. Note Maxima, minima, time period, and shape integrity
* Check Maxima minima of raw data. Since the sine waves are bipolar and readings are 10bit, result should be 512+x, 512-x . If that is not 0,1024 then there is no clipping issue due to PGA, and your problems lie elsewhere.
* Check Maxima minima of polynomial output. It's possible that the constant is not being applied since your plot is unipolar

viveksb007, 2017-07-01:
Got it working. [Screencast](https://goo.gl/photos/xyc6qWC7tyytF9QM9).
Does slider of **W1** on Hardware overrides frequency of W1 channel?
Could you check the screencast and tell how are things now?
Flow is still same 
setSine1(3000)
followed by captureTraces(1, 1000, 10, "CH1", trigger = false, CH123SA = 0)
delay
fetchTrace(1)

There was some error in Polynomial Initialization and unpacking calibration data. Data is read from device from LSB, so here is should be decoded as Little-Endian.
Little Endian : (byte[0] | (byte[1]<<8))
Big Endian :  (byte[0]<<8) | (byte[1])
[Reference](https://stackoverflow.com/questions/5616052/how-can-i-convert-a-4-byte-array-to-an-integer)
Could you tell if I am going in right direction?

jithinbp, 2017-07-01:
Cool. Slider on hardware is for amplitude control.

CloudyPadmal, 2018-06-21:
Fixed in #1069

PR #257 (2017-07-02): Improved readback speed and fixed various bugs
Fixes issue #255 #256 
Partially Fixes #249 

Changes: 
- fixed calibration data unpacking 
- byte order in making int from bytes
- fixed unsigned problem of byte ( In JAVA unsigned byte isn't supported, casted byte as ```int```, then ```ANDed``` with ```0xff```  )
- Fixed ```Polynomial Function``` Creation, unlike numpy Apache Common Math calculates value at a data point with reverse order of coefficient w.r.t ```numpy1D``` function.  

There are still readback error but now we can read most of sample data.

[Screencast](https://goo.gl/photos/ANi [... cut]
closed 2018-06-21 by CloudyPadmal