fossasia/pslab-app #251: UI for Control is buggy

viveksb007, 2017-06-29:
**Actual Behaviour**

Non-Uniform UI for Control Main

**Expected Behaviour**

Should be uniform for all screens.

**Steps to reproduce it**

Open Control Main

**LogCat for the issue**

NA

**Screenshots of the issue**

[image]

[image]

**Would you like to work on the issue?**

@asitava1998 @CloudyPadmal I think you guys should re-think about your approach to make UI for Control. No need to put everything in same line. You can put everything beautifully in Card-Type View with proper padding of individual elements.

asitava1998, 2017-06-29:
@viveksb007  @CloudyPadmal The first menu was not like that. Padmal, this happened after you created a separate UI for different screens, so please have a look at what went wrong there.
As of cards, I was thinking to do it like Science Journal did it.  Each separate feature gets a separate card.

viveksb007, 2017-06-29:
I think its about time we start writing tests too. I don't know so I would have to learn it too.
And from now try to provide an apk for testing with PR and at least 2 devs should test that apk before the PR is merged.
And @CloudyPadmal Please have a look at the issue.

asitava1998, 2017-06-30:
Yes. That would a good idea to write tests. 
Another thing we could do to know more about the UI is somehow learn how the open-event and phimpme guys did it. They have a good UI and maybe knowing how  it was developed may help a bit.

CloudyPadmal, 2017-06-30:
Sure @viveksb007 and @asitava1998 ! I'll look into the UI issue in main control panel now! :+1:

viveksb007, 2017-06-30:
@asitava1998 Their UI also improved over time. We should not condense a lot of view elements together as there are different dimensions in mobile screens. 
@CloudyPadmal Thanks

commit 8e7baaf2 (2017-07-02): Partially fixes #251 : Main Control UI (#254)

* Partially fixes #251 : Main Control UI

* Advanced control UI panel modification
closed 2017-07-02 by viveksb007