fossasia/pslab-app #262: Complete remaining portion of Sensor Activity

asitava1998, 2017-07-02:
**Actual Behaviour**

Presently the `AutoScan` button of Sensor QuickView works and the results of AutoScan are displayed. However, the list of available sensors are not reflected under `Select Sensor` and as a result `Get raw` is not functional.

**Expected Behaviour**

Both the `Select Sensor` spinner and `Get Raw` button must be available.

**Steps to reproduce it**

Main Menu -> Applications -> Sensor Quickview -> Click on `AutoScan` -> The result will be displayed in the test-box below but the spinner & button below it are not functional.

**LogCat for the issue**

None

**Screenshots of the issue**

None

**Would you like to work on the issue?**

Yes

viveksb007, 2017-07-03:
@asitava1998 Could you tell us how you are approaching this? and How do you plan to expose sensor data to user?

asitava1998, 2017-07-03:
@viveksb007  As of now after initial discussion with Jithin, I first made a temporary UI which will detect the sensor fetch the raw data by calling the method `getRaw`. Everything till now happens on one window.

However, the plan is as we land on the `Sensor QuickView` page, there is an `AutoScan` functionality. The list of available sensors is displayed on clicking AutoScan. The list of sensors is appended in a spinner and as we select an item in a spinner, we land on the UI of that specific sensor.

However, this PR is just to complete the Sensor Activity in the temporary UI.

viveksb007, 2017-07-03:
Okay By clicking On Auto Scan we get address on which sensor is attached right? How do we know which sensor it is?
And I was thinking after Auto Scan we populate a List and on clicking on listItem, a fragment corresponding to that sensor (ListItem) would come up with its UI to show its relevant data.

asitava1998, 2017-07-03:
Clicking `AutoScan` returns an ArrayList of integers and each integer corresponds to a unique sensor. So, we know which sensor it is as the numbers are unique.
About the second part to call the fragment corresponding to that sensor, I am planning to do the same.

commit 84c7f33e (2017-07-07): Fixes #262 completes remaining portion of Sensor Activity (#271)
closed 2017-07-07 by viveksb007