fossasia/pslab-app #272: Add Toolbar in LogicalAnalyzerActivity

nikit19, 2017-07-04:
**Actual Behaviour**

There is no toolbar in LogicalAnalyzerActivity

**Expected Behaviour**

A toolbar with a back button and name of the activity should be present

**Steps to reproduce it**

Select the Applications fragment and click on Logical Analyzer.

**LogCat for the issue**

.

**Screenshots of the issue**

Before
[image]

After
[image]

**Would you like to work on the issue?**

Yes I would like to work on this issue.

viveksb007, 2017-07-04:
Hi @nikit19,
Send a PR for same. Thanks

viveksb007, 2017-07-04:
Fixed in #275

closed 2017-07-04 by viveksb007