fossasia/pslab-app #279: Create sensor specific UI - HMC5883L & MPU6050

asitava1998, 2017-07-05:
**Actual Behaviour**
No sensor specific UI till now. Only a common UI to get raw data.

**Expected Behaviour**
Each sensor should have its own UI for displaying data. 

- [ ] Create mock-ups for UI.
- [ ] Create the layout files.
- [ ] Add functionality

**Steps to reproduce it**
Main Menu -> Applications -> Sensor Quickview

**LogCat for the issue**

None

**Screenshots of the issue**

None

**Would you like to work on the issue?**

Yes.

asitava1998, 2017-07-06:
The mock-ups for the UIs are [here](https://app.moqups.com/asitavasarkar/Wwr6F3xmaB/view/page/a0def9742).

akarshan96, 2017-07-06:
@asitava1998 great work 👍

CloudyPadmal, 2017-07-06:
Nice work @asitava1998 ! :+1:

asitava1998, 2017-07-10:
@akarshan96 @CloudyPadmal @viveksb007 @jithinbp 
This is what the layout files roughly look like.
[image]
[image]

The fragment files are also done and changes in layout files can be made to make it look better in subsequent issues.

What should be the transition from Sensor QuickView menu to each of the fragment? Vivek had suggested that we list all the connected sensors in a ListView and on clicking any one of the items we will move to the fragment of that particular sensor. Any other ideas regarding the same or the ListView is final? I have to make changes accordingly.

akarshan96, 2017-07-10:
@asitava1998 layout looks good to me. You can also try this [link](https://github.com/cachapa/ExpandableLayout), see if this solves the purpose, look for "Accordion" expansion.

viveksb007, 2017-07-10:
Looks great.
Graph under Plot Section could be made collapseable.

asitava1998, 2017-07-10:
@akarshan96 @viveksb007 What about the transition from activity to each fragment? When `AutoScan` is clicked, the sensors available are displayed, so should they be arranged in a listview, accordion etc. and on clicking we move to the selected fragment?

viveksb007, 2017-07-10:
ListView is okay for now. Rest if you have any better option in mind go ahead.

CloudyPadmal, 2017-07-10:
So all the sensors are listed in a ListView and when clicked upon, each will open up a new fragment right?

asitava1998, 2017-07-10:
@CloudyPadmal That's the initial thought. If you have a better idea, let me know.

viveksb007, 2017-07-16:
Fixed in #295

closed 2017-07-16 by viveksb007