fossasia/pslab-app #286: App crash on setting Voltage/Current  from Control Main

viveksb007, 2017-07-07:
**Actual Behaviour**

When **SET** button is pressed and device is connected, App crashes. 

**Expected Behaviour**

App shouldn't crash instead whatever voltage/current is specified in EditText should be set. 

**Steps to reproduce it**

Connect Device and try to set Voltage/Current. 

**LogCat for the issue**

NA, Activity is restarted so error Logs aren't captured.

**Screenshots of the issue**

NA

**Would you like to work on the issue?**

Yes

viveksb007, 2017-07-07:
@asitava1998 Did you tested PVX, PCS with device after adding setVoltage / setCurrent calls ?

asitava1998, 2017-07-07:
No, I hadn't tested it. I left it commented but few days back someone (can't remember who did) uncommented and slightly modified some of the code in Control Main, so didn't test it.

viveksb007, 2017-07-07:
No problem. I would debug it. But from next time review PR carefully and raise question if you find something that can potentially result in app crash.

commit 850c7719 (2017-07-08): fixed crash on setting voltage/current and added left-out callbacks (#288)

* fixed crash on setting voltage/current and added leftout callbacks

* removed unused imports
Fixes #286 
Fixes #287
closed 2017-07-08 by viveksb007