fossasia/pslab-app #292: Create UI for performing experiment listed in Saved Experiment Fragment

viveksb007, 2017-07-09:
**Actual Behaviour**

List of Experiment is shown in Saved Experiment Fragment.

**Expected Behaviour**

On clicking any experiment from that list corresponding Experiment Fragment/Activity should open.

**Steps to reproduce it**

NavDrawer->Saved Experiment

**LogCat for the issue**

NA

**Screenshots of the issue**

NA

**Would you like to work on the issue?**

Yes

viveksb007, 2017-07-10:
@jithinbp @akarshan96 @asitava1998 @CloudyPadmal 
Could you guys have a look and give your feedback/suggestions on layout?
Screencast Link : https://goo.gl/photos/LKLXjz9QkVuuFZnY7
 
Lower part of first image is completely different layout which would vary with experiment. In upper part only Image and Title would vary. Screencast has shown this better.
On clicking play button, one fragment would come over which would have view components to perform experiment.

<table>
<tr>
<td>[image]</td>
<td>[image]</td>
</tr>
</table>

jithinbp, 2017-07-10:
Cool. It feels quite elegant.

I have questions/suggestions:
* When you hit the play button, is there going to be an easy way to switch between the doc screen and the experiment activity without having to reconfigure the parameters?  It would be nice if the two would work together with no need to use the 'back' button.
* In the main screen subtitles, we can maybe do away with the 'Experiments related to ...' text . It's probably obvious
* Can you reuse the docs from pslab-desktop-apps?  They use pictures from designated subfolders, so all you have to do is replace the 'screenshots' folder with app screenshots , and recompile the static site. Will save effort on the documentation front since the text, schematics and photographs are reusable.

thanks

viveksb007, 2017-07-10:
- We can have something like YouTube, when experiment fragment could be compressed in lower right corner like YouTube videos and we can inspect the doc like we can scroll for videos in youtube.
- yeah we can remove the subtitles, would do that in next PR.
- about re-using the pslab-desktop-app docs, we can do that but putting all that html files in app itself would increase size of app. The size problem is with current implementation too. It would be great if we can host those files somewhere and I can request them in web-view in Android. If size isn't a issue currently, I can give it a try ( re-using pslab-desktop-app docs ).

jithinbp, 2017-07-10:
Sounds like a lot of coding effort, but also sounds like great UX to be able to do that.  We may have to be careful with the swipe-down-to-view-docs , and instead provide a button or something. Since it's not a video player, I believe the swipe action will be used by graphs in experiments.

It could be a separate data download, since google play limits APK sizes in my opinion. For example, most games download resources separately.
I'm not in favour of live hosting, because internet cannot be taken for granted in most places

thanks

viveksb007, 2017-07-10:
Okay then I would try to use pslab-desktop-app docs in lower section of first Image shown above. And for now I would overlay another fragment on clicking play button. Once experiment functionality is done. I would look for compressing it to lower corner on clicking some button. 
Jekyll generated HTML files in pslab-desktop-app handles resizing on mobile device on its own. Right?

jithinbp, 2017-07-10:
sounds good. You could  simply copy paste one of the doc files and its images and see how it behaves in the WebView in android.
I've used bootstrap , so it should resize well. looks okay on my phone. If there are any problems, we can always modify the css file.

CloudyPadmal, 2017-07-11:
Nice work @viveksb007 ! :100: Yes like @asitava1998 mentioned, Once we click on a sub list item and go into that specific experiment UI, if we can swipe left and right to get the other items in the sub list it would be nice! :+1:

viveksb007, 2017-07-16:
Fixed in #303 #312 #314

closed 2017-07-16 by viveksb007