fossasia/pslab-app #302: Buttons in 'Sensor QuickView' do not implement selectors

CloudyPadmal, 2017-07-12:
**Actual Behaviour**

Buttons in `Sensor QuickView` do not implement selectors.

**Expected Behaviour**

Buttons should have an effect on clicking to show that the button is pressed.

**Steps to reproduce it**

1. Open app
2. Go to `Applications`
3. Choose `Sensor QuickView`
4. Click the buttons and observe

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

asitava1998, 2017-07-12:
Don't do it for the `Get raw` button though, it is being removed anyway.

CloudyPadmal, 2017-07-12:
Sure @asitava1998 ! :+1:

commit 704d95f6 (2017-07-12): Fixes #302 : Buttons in Sensor Quickview fix (#307)
closed 2017-07-12 by viveksb007