fossasia/pslab-app #339: Add more functionality to HMC5883L sensor

akarshan96, 2017-07-18:
**Actual Behaviour**

 HMC5883L sensor fragment doesn't have all the functionalities.

**Expected Behaviour**

HMC5883L sensor fragment should have

- [ ] Pause/record button
- [ ] Time gap between each data point option
- [ ] No. of data points to be recorded option
- [ ] Trigger function

**Steps to reproduce it**

Menu Drawer -> Applications -> Sensor QuickView -> HMC5883L 

**LogCat for the issue**

> None

**Screenshots of the issue**

> None

**Would you like to work on the issue?**

Yes

akarshan96, 2017-07-23:
@jithinbp @asitava1998 @viveksb007 @CloudyPadmal, I am sharing a mockup for control bar in Sensor activity. I have added 

1. a common play and pause button. 
2. a seek-bar to determine the time gap between each data point capture.
3. an edit-text box in which no. of samples to be captured can be entered.
4. a checkbox when checked will capture data indefinitely. This will disable the edit-text box.

Please review it and give suggestions.

[image]

viveksb007, 2017-07-23:
@akarshan96 Looks great. Instead of using TextView and EditText for taking Number of samples use ```TextInputEditText``` wrapped inside ```TextInputLayout```. Refer [this](https://github.com/fossasia/pslab-android/blob/master/app/src/main/res/layout/diode_configure_dialog.xml#L17).

viveksb007, 2017-08-29:
@akarshan96 Please update the Issue description about what has been achieved and if anything is remaining, please open a sub-issue.

PR #595 (2017-08-29): Added functionality to remaining Sensor fragments
Fixes #339

Changes: Added functionality to remaining Sensor fragments

Screenshots for the change: None
commit afb708b3 (2017-08-29): Added functionality to remaining Sensor fragments (#595)

* Adds functionality to remaining sensor fragments

Fixes #339
closed 2017-08-29 by viveksb007