fossasia/pslab-app #353: Oscilloscope indicator LED and Graph don't get updated

CloudyPadmal, 2017-07-21:
**Actual Behaviour**

When oscilloscope is opened while the device is connected, incoming signal is not displayed unless I deselect a checkbox and select it again

**Expected Behaviour**

When the oscilloscope is open, the waveform needs to be appear in the graph view. It was functional before.

**Steps to reproduce it**

1. Open app and connect PSLab device
2. Go to `Applications` and `Oscilloscope`
3. Observe the LED is not lit and try check a checkbox or deselect CH1 and select it again

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[Recording](https://drive.google.com/open?id=0B5su2_aUn8g7MUstdklPMERRZDA)

**Would you like to work on the issue?**

I would like to fix this!

viveksb007, 2017-07-21:
@CloudyPadmal we don't want to show waveform right away as the oscilliscope activity is opened. @akarshan96 disabled it but probably forgot to uncheck it from the xml files. So just uncheck it from xml files only.

CloudyPadmal, 2017-07-21:
In this PR [#309](https://github.com/fossasia/pslab-android/pull/309) 8 days ago it is specifically enabled. Wouldn't it be convenient for a user to have a default channel?

viveksb007, 2017-07-21:
Nope actually during adding in-built mic feature it was hindering. We can do it when releasing for user. In the development phase it doesn't make much sense.

CloudyPadmal, 2017-07-21:
Ahh so by default all the check boxes will be empty right? I'll make changes to the xml files and send a pull request! :+1:

commit e9384133 (2017-07-21): Fix #353 : Removed default selection CH1 in OSC (#355)
closed 2017-07-21 by viveksb007