fossasia/pslab-app #363: create DataSensorLogger activity with required runtime permission

viveksb007, 2017-07-23:
**Actual Behaviour**

Currently there is nothing in DataSensorLogger tile in Application Fragment.

**Expected Behaviour**

DataSensorLogger Activity should start on clicking DataSensorLogger tile in Application Fragment. Functionality of DataSensorLogger Activity:
- [ ] choose a sensor and start logging data and export it as excel (or any other appropriate format).
- [ ] Enable logging as a background service. Notification (with play, pause, stop button) would come as soon as logging start and we can do anything else on mobile. We can stop logging data by clicking on stop button on App Notification on Notification bar. (Similar to how we play, pause music).     

**Steps to reproduce it**

NA

**LogCat for the issue**

NA

**Screenshots of the issue**

NA

**Would you like to work on the issue?**

Yes

viveksb007, 2017-07-23:
@jithinbp @akarshan96 @asitava1998 @CloudyPadmal Please review the **Expected Behaviour** of DataSensorLogger Activity and any addition or suggestion is welcome.

asitava1998, 2017-07-23:
@viveksb007 The second point sounds good. Regarding the first point, when the logging is finished, a page is displayed with some statistics like average value, standard deviation and maybe a graph to display the logged data and an option to export it has csv, excel etc.

CloudyPadmal, 2018-07-16:
Sensors are implemented as separate instruments

closed 2018-07-16 by CloudyPadmal