fossasia/pslab-app #446: Implement Half Wave Rectifier Experiment

akarshan96, 2017-08-06:
**Actual Behaviour**

Half Wave Rectifier Experiment is not yet implemented.

**Expected Behaviour**

Half Wave Rectifier Experiment needs to be implemented

**Steps to reproduce it**

1. Open app
2. Go to `Saved Experiments`
3. Select `Electronics Experiments`
4. Select `Diode IV Characteristics`
5. Select `Half Wave Rectifier`

**LogCat for the issue**

> None

**Screenshots of the issue**

> None

**Would you like to work on the issue?**

Yes

akarshan96, 2017-08-08:
@viveksb007 @asitava1998 @CloudyPadmal for implementing Half Wave Rectifier Experiment, I have reused Oscilloscope Activity. Sharing the screenshot. Please give suggestions on this.
[image]

CloudyPadmal, 2017-08-08:
This will be great! :+1:

viveksb007, 2017-08-08:
Looks good.
We don't need right Y-axis.

jithinbp, 2017-08-08:
@viveksb007 There will be two plots (CH1 : complete wave,  CH2 : rectified ) . If you remove the right axis, the left axis label should not be simply 'CH1 (V)' .

We can also simplify this experiment by removing timebase and frequency configuration options. Simply use a fixed frequency, and display it prominently at the bottom. It's just a suggestion. For beginners, too much flexibility can be confusing.

akarshan96, 2017-08-08:
@jithinbp made the suggested changes. Sharing the screenshots. @viveksb007 @asitava1998 @CloudyPadmal please give your reviews too.

[image]

akarshan96, 2017-08-08:
I have made the suggested changes. @jithinbp @viveksb007 @asitava1998 @CloudyPadmal please review it.
[image]

jithinbp, 2017-08-08:
Is there some issue with CH2 on your device? It shouldn't be at ~6V unless you've connected something to it. Can you use a diode from the accessory kit to make a screenshot of the halfwave experiment?

PR #468 (2017-08-10): Implemented Halfwave Rectifier Experiment
Fixes #446 
Changes: 
Implemented Halfwave Rectifier Experiment

Apk:
[app-debug.apk.zip](https://github.com/fossasia/pslab-android/files/1209660/app-debug.apk.zip)
*In the screenshot I have mapped output signal to CH1 for the demo purpose. Please check the apk for both input and output signals.
 
Screenshots for the change: 
[image]
commit c0081da8 (2017-08-10): Implemented Halfwave Rectifier Experiment (#468)

* Implements Halfwave Rectifier Experiment

* Made suggested changes
Fixes #446
closed 2017-08-10 by viveksb007