fossasia/pslab-app #457: sensor list shows repeated sensors on multiple scans

viveksb007, 2017-08-07:
**Actual Behaviour**

sensor list shows repeated sensors on multiple scans

**Expected Behaviour**

Sensor list should show only sensors connected i.e previous data shouldn't be showed to user.

**Steps to reproduce it**

Scan Sensor by clicking on **Auto Scan** again and again.

**LogCat for the issue**

NA

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

hardik124, 2017-08-07:
@viveksb007  Can I work on this issue?

viveksb007, 2017-08-08:
@hardik124 sure I haven't started working on it. So you can go ahead to solve this issue.
Please go through the guidelines once to begin contribution https://docs.google.com/document/d/1E4kNgrOSw64R2IAG45JSIgQL9Yxia-mG2dqlRcrUo-U/edit
And be quick to send a PR as its a bug.
Thanks.

hardik124, 2017-08-08:
@viveksb007  , i dont have a device, any way to emulate it ?

hardik124, 2017-08-08:
@viveksb007  , i think the problem was that clear() was not called on array lists , hence it made copies of data , 
So i called clear on both arrays , although i have no way to check it , since i dont have a device .

Also , do you think ,we can move the code that get all sensors in another method ad call it on onCreate and onClick to avoid code repetition. ?

viveksb007, 2017-08-08:
Yes scanning sensors and adding it to array can be put inside a function and called whenever needed. After you are done with changes send a PR with the apk. I will check it.

viveksb007, 2017-08-10:
Fixed in #467

closed 2017-08-10 by viveksb007