fossasia/pslab-app #521: Implement a uniform button in saved experiments

CloudyPadmal, 2017-08-15:
**Actual Behaviour**

The application uses a selector for its button background in control UI, Oscilloscope and Data sensor logger. But in saved experiment, stock button is used which is gray in color

**Expected Behaviour**

App needs to implement a uniform button everywhere in the application

**Steps to reproduce it**

Observe experiments related buttons in `Saved Experiments`

**LogCat for the issue**

> Not available

**Screenshots of the issue**

| Other Screen | Saved Experiments |
| ---- | ---- |
| [image] | [image] |

**Would you like to work on the issue?**

Yes

viveksb007, 2017-08-15:
@CloudyPadmal stock button looks better in my opinion.
Making all buttons ```RED``` doesn't make the app look good.

CloudyPadmal, 2017-08-16:
Hi @viveksb007 ! My suggestion is to implement the same theme through out the app. What do you think? :)

vikasmahato, 2017-10-12:
@CloudyPadmal   @viveksb007 Can I work on this one??

CloudyPadmal, 2017-10-12:
Sure! :+1:

sharmamohit123, 2018-02-15:
@CloudyPadmal  I also want to work on this issue.

CloudyPadmal, 2018-02-16:
Sure go ahead @sharmamohit123 ! 👍

CloudyPadmal, 2018-06-18:
Fixed in #714

closed 2018-06-18 by CloudyPadmal